f4exb/sdrangel #1482: Feature: write sigmf files in other formats

catkira, 2022-10-17:
SDRangel currently writes sigmf files in ci32 format. However GNU Radio and inspectrum dont support this format (https://github.com/miek/inspectrum/issues/209). Is it possible to make an option in SDRangel to write other formats?

catkira, 2022-10-17:
yes, but that also does not work with inspectrum.

srcejon, 2022-10-17:
Even if you rename it to .s16 or .sc16?

f4exb, 2022-10-18:
And among the zillions of formats it already supports: https://github.com/miek/inspectrum#input it is still to complicated for them to add ci32 which is a valid sigmf type?

catkira, 2022-10-18:
It seems so. Somebody offered a PR to support ci32, but there was no interest (no reply) from miek.
https://github.com/miek/inspectrum/issues/209#issuecomment-1281374150

catkira, 2022-10-18:
sigmf in GNU radio also does not support the ciXX format 
https://github.com/gnuradio/SigMF/blob/sigmf-v1.x/sigmf/sigmffile.py#:~:text=is_unsigned_data%20%3D%20%22u,not%20in%20datatype

catkira, 2022-10-18:
sigmf in GNU radio also does not support the ciXX format 
see [here](https://github.com/gnuradio/SigMF/blob/sigmf-v1.x/sigmf/sigmffile.py#:~:text=is_unsigned_data%20%3D%20%22u,not%20in%20datatype)

f4exb, 2022-10-18:
I don't know why people consider fixed point arithmetic which results in data of (signed) integer type so weird. The reason this is used here is that it is a direct translation from internal I/Q samples representation and thus the most efficient. I suppose that ci32 corresponds to the case when SDRangel is compiled with 24 bit samples option (the default) and that it would be ci16 in the case it is compiled with the 16 bit option. It is always possible to add an optional transformation stage knowing that there will be a performance hit that may be detrimental for high throughput.

Normally it is up to the reading application to adapt to the data format (stated in the meta info) and not the writer. On the reading side this is exactly what the SIGMFFileInput plugin does: https://github.com/f4exb/sdrangel/tree/master/plugins/samplesource/sigmffileinput#4-sample-format

catkira, 2022-10-18:
writing as ci16 would also make the file a lot smaller.
But to me it seems the problem is that SigMF in gnuradio does not know the "i" type for signed integer. It just knows "u" for unsigned integer.
A fix for inspectrum would be straight forward, one would just have to add a `ComplexS32SampleAdapter` in this file, similar to the already present `ComplexS16SampleAdapter`.

catkira, 2022-10-18:
I made a PR to inspectrum that adds ci32 support: https://github.com/miek/inspectrum/pull/213

So changing SDRangel is not so urgent anymore. It would just be a consideration if one wants to reduce the file size.

catkira, 2022-10-19:
My PR for inspectrum got merged. So I think we can close this issue.

closed 2022-10-19 by catkira