f4exb/sdrangel #1484: Traking issue with rotator controller

Valstee, 2022-10-18:
Hi @f4exb, Hi @srcejon,
I use SDRangel 7.8.0, Qt 5.15.3 on Ubuntu 22.04 LTS.
I noticed a problem in the tracking function of Rotator Controller. During an AOS in the satellite tracker feature, the tracking (on rotator controller) does not start although the box provided for this purpose is checked. If I uncheck and then check the box the tracking start.
Do you have the same problem ?
I used the API to check the state of rotator controller and the "track" parameter is in state "1" but nothing happens until I change it to "0" and then back to "1".
Thanks, F4JIZ

Valstee, 2022-10-21:
Maybe that's why when I open Sdrangel although satellite tracker and rotator controller are open : 
`2022-10-21 10:32:55.683 (D) GS232Controller::applySettings: No plugin corresponding to source  "F0:0 SatelliteTracker"`

Valstee, 2022-10-24:
I used to launch SDRAngel and run several features via a python program including rotator controller. 
When SDRangel closes with the rotator controller tracking option enabled it was still enabled when I turned sdrangel back on.
Now it's not the case anymore when I restart sdrangel. 
The option is always checked, when we make a "GET parameters" the tracking is at 1 but it does not work. 
It just need to send a command like "PATCH "track = 1"" and that functions without problem. 
This may just be due to a change in the sequencing of the loading of features at startup !
Thanks F4JIZ

f4exb, 2022-10-24:
> When SDRangel closes with the rotator controller tracking option enabled it was still enabled when I turned sdrangel back on.
Now it's not the case anymore when I restart sdrangel.

Well it is... This is after re-launching SDRangel with the rotator controller linked to the start tracker:
[image]

srcejon, 2022-10-24:
I think he might mean the Track checkbox isn't checked. I can take a look in a few days - away at the moment.

Valstee, 2022-10-25:
Yes, exactly.
Step 1: I check the tracking box. Everything works fine. 
2nd step : I close sdrangel (keeping the tracking option checked).
3rd step : I reopen sdrangel, the features are there, the tracking option is checked but it doesn't work. I have to uncheck and recheck the tracking option for it to really work.

srcejon, 2022-10-27:
> Well it is... This is after re-launching SDRangel with the rotator controller linked to the start tracker: [image]

The list of sources is populated when the Rotator is F:0 and Star Tracker is F:1, but not when the other way around (i.e. when Rotator Controller is F:1, I guess it doesn't get the MainCore::featureAdded added signal, and so GS232Controller::notifyUpdate() isn't called).

srcejon, 2022-10-28:
I can't reproduce the original reported problem though. I can see commands sent to the rotator controller on AOS, having only pressed the start buttons in the GUI.

Perhaps you could send a screenshot, so I can see other setup/settings that might interfere.

closed 2023-01-03 by github-actions[bot]