f4exb/sdrangel #1485: Segmentation Fault on open

CRD716, 2022-10-23:
gdb output:
`2022-10-23 18:34:05.186 (D) DeviceUISet::loadTxChannelSettings: deserializing channel [sdrangel.channeltx.modssb]
2022-10-23 18:34:05.187 (D) GLSpectrumGUI::applySettings: refLevel: 0.000000e+00 powerRange: 1.000000e+02
2022-10-23 18:34:05.187 (D) ScaleEngine::setTruncateMode: off
2022-10-23 18:34:05.187 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrumView::MsgReportCalibrationShift
2022-10-23 18:34:05.187 (D) SpectrumVis::handleMessage: MsgConfigureSpectrumVis
2022-10-23 18:34:05.187 (D) SpectrumVis::applySettings:  m_fftSize:  1024  m_fftWindow:  4  m_fftOverlap:  0  m_averagingIndex:  0  m_averagingMode:  0  m_refLevel:  0  m_powerRange:  100  m_fpsPeriodMs:  50  m_linear:  false  m_ssb:  false  m_usb:  true  m_wsSpectrumAddress:  "127.0.0.1"  m_wsSpectrumPort:  8887  force:  false
2022-10-23 18:34:05.187 (D) CWKeyerGUI::on_keyboardKeyer_toggled: true

Thread 1 "sdrangel" received signal SIGSEGV, Segmentation fault.
0x00007ffff5ee4afa in QObject::connect(QObject const*, char const*, QObject const*, char const*, Qt::ConnectionType) () from /lib/x86_64-linux-gnu/libQt5Core.so.5`

Latest version (and a few in the past)
Running on Debian with KDE
I've tried updating, building again, among other things.
At this point I'm completely out of ideas.

CRD716, 2022-10-25:
If more information is needed I'm happy to give it.

CRD716, 2022-10-25:
[Here's a full gdb output if it helps](https://drive.google.com/file/d/1EjY8JdA0dGI9giJbImVHyLAiuC2NcHOA/view?usp=sharing)

f4exb, 2022-10-25:
The stack trace should be more helpful you can obtain it also via gdb and with the core dump. The command is something like: `gdb -c core.12345 ./sdrangel` (12345 is the process id). Although the log shows you are using an Audio Output device with a SSB modulator. 

I was able to reproduce and understand what happens with a similar context using the AM modulator that also has the CW keyer. In fact when you first create the modulator there is no issue because the main window is already there but the CommandKeyReceiver is created after the presets are loaded so when you start again it fails on the connection because its address is not valid. https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L228

So it should be created at an earlier stage before the default preset is loaded.

commit 9b7a41ad (2022-10-25): MainWindow: move CommandKeyReceiver at an earlier stage in the constructor. Fixes #1485
closed 2022-10-25 by f4exb