f4exb/sdrangel #1486: How to install sdrangel on Imac i5 macOS High Sierra?

snwvo7, 2022-10-24:
Is there a way to install a version of sdrangel on a Imac i5 macOS High Sierra?
I am unable to compile using (Xcode 11.3.1) due to incompatible wit High Sierra.
Is there a DMG available?

CRD716, 2022-10-24:
The last DMG is a version behind, but should work until a new version is made. 
https://github.com/f4exb/sdrangel/releases/download/v7.8.0/sdrangel-7.8.0-macx64.dmg

snwvo7, 2022-10-24:
> 

Thanks for the quick response, this version only works on macOS 10.14 or higher. The version of macOS I am using is 10.13.6 is there an older version available?

CRD716, 2022-10-24:
I'm not sure which versions support which, but you can go through https://github.com/f4exb/sdrangel/releases and keep trying older and older versions. Sorry that this isn't an ideal solution, but I don't have an available mac myself.

srcejon, 2022-10-24:
There isn't a High Sierra DMG.

Can't recall why I suggested Xcode 11.3.1 is needed. Try building with whatever version of Xcode you have, then post any problems here, and I might be able to suggest something.

cygeus, 2022-10-25:
I don't have a mac running High Sierra any longer, but when I still did I compiled [SDRangel](https://ports.macports.org/port/SDRangel/details/) myself using [MacPorts](https://www.macports.org/).

As a side note: High Sierra is quite old nowadays. Not supported by Apple (and subsequently other developers) for a few years now. So missing out on security updates and newer versions of several programs don't work.
If possible, I advise you to look into [dosdude1 patches](https://dosdude1.com/software.html) to upgrade to Mojave (10.14) or Catalina (10.15). Or even better into [OpenCore Legacy Patcher](https://dortania.github.io/OpenCore-Legacy-Patcher/) to upgrade to Big Sur (11) or Monterey (12). I'm not affiliated with any of these projects, just a happy user of them.

closed 2022-12-30 by github-actions[bot]

fehmi, 2024-08-22:
> The last DMG is a version behind, but should work until a new version is made. https://github.com/f4exb/sdrangel/releases/download/v7.8.0/sdrangel-7.8.0-macx64.dmg

the link is broken.

CRD716, 2024-08-24:
> > The last DMG is a version behind, but should work until a new version is made. https://github.com/f4exb/sdrangel/releases/download/v7.8.0/sdrangel-7.8.0-macx64.dmg
> 
> the link is broken.

Does the latest version at https://github.com/f4exb/sdrangel/releases/download/v7.22.0/sdrangel-7.22.0-macx64.dmg not work for you?

fehmi, 2024-09-01:
> Does the latest version at https://github.com/f4exb/sdrangel/releases/download/v7.22.0/sdrangel-7.22.0-macx64.dmg not work for you?

Didn't work. It says "You have macOS 10.13.6. The application requires macOS 10.14 or later."