f4exb/sdrangel #1496: Spectrum max hold / decay 0

srcejon, 2022-11-04:
In the docs for the spectrum decay dial, it says:

"A value of zero means no decay and thus phosphor history and max hold (red line) will be kept until the clear button (B.1.4) is pressed."

However, when I set decay to 0, max hold seems to be reset when the mouse is moved over a channel window, if the waterfall isn't being displayed.

To recreate:

- Enable max hold
- Set decay to 0
- Add a channel
- Disable waterfall (so only spectrum is displayed)
- Move mouse between spectrum and channel

f4exb, 2022-11-04:
it happens because of this line of code https://github.com/f4exb/sdrangel/blob/master/sdrgui/gui/glspectrumview.cpp#L3306 thus the histogram data is systematically cleared here https://github.com/f4exb/sdrangel/blob/master/sdrgui/gui/glspectrumview.cpp#L3354 unless `fftSizeChanged` is properly set which happens only if the waterfall is displayed. 

Note that this happens each time `applyChanges()` is called which happens when `m_changesPending` is set to true and happens in other occasions like spectrum window resizing ...

f4exb, 2022-11-04:
I don´t see a way to fix this

commit f5b91fff (2022-11-04): GLSpectrum: separate waterfall and histogram resizing conditions. Fixes #1496
closed 2022-11-04 by f4exb