f4exb/sdrangel #1504: New Feature -  9600 PSK packet (de)modulator

N5BRG, 2022-11-07:
I am looking for a way to decode the LEO satellites which use 9600 PSK for beacon and packet operation.

If there is a way to do this now I am not finding it?

I have looked at the code but I am thinking this wil be too difficult for me to make changes due to the number of cpp files involved.

Thanks

srcejon, 2022-11-07:
Not currently. Which specific satellite are you trying to receive.

N5BRG, 2022-11-07:
Falconsat-3
https://www.amsat.org/falconsat-3/#:~:text=Modulation%20is%209600%20bps%20GMSK%20for%20the%20uplink,performed%20remarkably%20well%20for%2010%20years%20on%20orbit.

This satellite is decaying and will be gone by March of 2022. Currently being turned on during weekends.

Greencube
https://www.s5lab.space/index.php/greencube-home/

Also working with a small AMSAT NA team look at a design for Packet which may use 9600 PSK downlink and I wanted to use SDRAngel to get some experience with these Satellites.

srcejon, 2022-11-07:
I do want to support more things like this, but probably not going to be in the immediate future, as working on a few other features.

Falconsat-3 doesn't come over the UK, so for that, it would be useful if you could upload an IQ capture to this issue.

Greencube can't seem to be tracked with the satellite tracker for some reason, so a possible bug there. Again, if you have an IQ file, it would be helpful.

N5BRG, 2022-11-07:
Thank you for your feedback and comments.

I will get a IQ capture and upload it to the issue.

N5BRG, 2022-11-07:
I am able to see the track for Greencube on SDRAngel 7.8.1-52 64 bit build if that helps to understand the issue there.
My build is a Docker build.

N5BRG, 2022-11-27:
I finally managed to capture some data.
FalconSat-3  9600 bps GMSK
GreenCube-3  modulation (https://www.s5lab.space/index.php/decoding-ledsat-2/)
I captured files with SDRAngel 7.8.2 20-gb6e13d1e6 Qt 5.15.3 Ubuntu 22.04.1 LTS
Screen capture of settings included.  I used desimation of 4 on File sink.
[image]
I see i have to make my capture files less than 25 MB. Will work on that and come back later.

N5BRG, 2022-11-27:
Falconsat- file
[FalconSat9b.2022-11-26T23_35_29_203.sdriq.zip](https://github.com/f4exb/sdrangel/files/10097453/FalconSat9b.2022-11-26T23_35_29_203.sdriq.zip)

N5BRG, 2022-11-27:
Greencube file
[Greencube4b.2022-11-26T02_57_47_429.sdriq.zip](https://github.com/f4exb/sdrangel/files/10097460/Greencube4b.2022-11-26T02_57_47_429.sdriq.zip)