f4exb/sdrangel #1513: Binary Build Request: Soapy plugin baked into Macos binary by default

dionmindhive, 2022-11-14:
Would be amazing to have the soapy SDR plugin/sample source enabled by default on the Macos (and I assume other OSs) binaries.
Not sure why it's not a default but have built from Macports - not there. 
Can see it as an option (disabled) in the full source build for macos but the environment is a mission to setup.
Any chance we could get that enabled in the precompiled binaries?
Would be great if possible.
Other than that, absolutely rad application. Hands down the best SDR frontend out there - would be that little bit better with default support for remote SDR instances :-)

dionmindhive, 2022-11-14:
Found (should have checked first) a macports variant. My bad (probably)

dionmindhive, 2022-11-15:
Still finding that the SDR module is disabled (though present) in that build:
2022-11-15 22:32:52.887 (I) PluginManager::loadPluginsDir: Soapy SDR disabled skipping libinputsoapysdr.dylib

Is this the expected behaviour?

srcejon, 2022-11-15:
I think so. You need to pass --soapy on the command line.

It's disabled by default, IIRC, as soapy can cause problems for some SDRs.

dionmindhive, 2022-11-15:
Ah - interesting. 
Is that in the docs somewhere? Had a good trawl but may have missed it

srcejon, 2022-11-16:
I thought there was at one point, but can't find it. If not, should probably be added to https://github.com/f4exb/sdrangel/wiki/Quick-start under "Invoke the program"

dionmindhive, 2022-11-16:
Probably a good idea. Frustratingly, the macports install seems to break SoapyRemote or SoapySDR such that I get no devices found errors when running SoapySDRUtil --find="remote=192.168.1.221:1234". Running port uninstall soapySDR cleans that up (after reinstalling soapyremote from Git) but breaks other things. Seems like port install sdrangel overwrites something important or removes the 'remote' factory. 
Would still just be awesome to have soapy support in the binary - with the ability to enable it with an arg if needed.

closed 2023-01-22 by github-actions[bot]