f4exb/sdrangel #1516: Bug APT demodulator channel

choupette-change, 2022-11-16:
Hello, APT demodulator channel no longer appears in the list of demodulators  in Ubuntu version ?
[image]
[image]

Valstee, 2022-11-18:
I think it's due to this durint compilation :
  ```
In file included from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.h:35,
                from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.cpp:19:
/opt/build/sdrangel/plugins/channelrx/demodapt/aptdemodimageworker.h:40:17: error: ‘libsgp4’ is not a namespace-name
40 | using namespace libsgp4;
      |                 ^~~~~~~
```

srcejon, 2022-11-18:
> I think it's due to that during the compilation :
> 
> ```
> In file included from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.h:35,
>               from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.cpp:19:
> /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemodimageworker.h:40:17: error: ‘libsgp4’ is not a namespace-name
> 40 | using namespace libsgp4;
>     |                 ^~~~~~~
> ```

This error means you need to update and rebuild sgp4 - and also delete CMakeCache.txt in your sdrangel build directory and re-run cmake.

choupette-change, 2022-11-18:
OK for me :) :)

closed 2023-01-24 by github-actions[bot]