f4exb/sdrangel #1517: libsgp4 error

lrowen, 2022-11-16:
Updated libsgp4 prior to rebuilding current sdrangel.  Relevant information:

System: Debian 11

```
lowen@lowen-7740:/opt/build/sgp4/build$ git show -s
commit 6a448b4850e5fbf8c1ca03bb5f6013a9fdc1fd91 (HEAD -> master, origin/master, origin/HEAD)
Author: Daniel Warner <dan.r.warner@gmail.com>
Date:   Sun Nov 13 13:22:13 2022 +0000

    Misc c++ updates (#30)
    
    * Misc c++ updates
    
    * update trim
    
    Co-authored-by: Dan Warner <info@danrw.com>
lowen@lowen-7740:/opt/build/sgp4/build$ 

```
lowen@lowen-7740:/opt/build/sdrangel/build$ git show -s
commit 006e23d7b729a6239d3599cfb0a0cc6bd84162db (HEAD -> master, tag: v7.8.3, origin/master, origin/HEAD)
Author: f4exb <f4exb06@gmail.com>
Date:   Tue Nov 15 21:46:52 2022 +0100

    Updated versions and changelogs
lowen@lowen-7740:/opt/build/sdrangel/build$ 

```

Error output:
```
...
In file included from /opt/build/sdrangel/build/pluginssrv/channelrx/demodapt/demodaptsrv_autogen/EWIEGA46WW/moc_aptdemodimageworker.cpp:10,
                 from /opt/build/sdrangel/build/pluginssrv/channelrx/demodapt/demodaptsrv_autogen/mocs_compilation.cpp:3:
/opt/build/sdrangel/build/pluginssrv/channelrx/demodapt/demodaptsrv_autogen/EWIEGA46WW/../../../../../../plugins/channelrx/demodapt/aptdemodimageworker.h:40:17: error: ‘libsgp4’ is not a namespace-name
   40 | using namespace libsgp4;
      |                 ^~~~~~~
[ 41%] Building CXX object pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodbaseband.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodsink.cpp.o
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/build.make:82: pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/demodaptsrv_autogen/mocs_compilation.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 41%] Building CXX object pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodplugin.cpp.o
In file included from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.h:35,
                 from /opt/build/sdrangel/plugins/channelrx/demodapt/aptdemod.cpp:19:
/opt/build/sdrangel/plugins/channelrx/demodapt/aptdemodimageworker.h:40:17: error: ‘libsgp4’ is not a namespace-name
   40 | using namespace libsgp4;
      |                 ^~~~~~~
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/tvscreen.cpp.o
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/build.make:95: pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemod.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:14951: pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

srcejon, 2022-11-16:
I think you may need to delete cmakecache.txt and rerun cmake.

Or build from a clean directory.

lrowen, 2022-11-16:
Ok, I'll give that a whirl tomorrow; thanks.

lrowen, 2022-11-17:
I'm old-school; I'll always do a 'make clean' in the build directory, but cmake reacts a bit differently... I guess building in a clean build directory needs to become my standard procedure.  This fixed the build; thanks!

closed 2022-11-17 by lrowen

srcejon, 2022-11-17:
make clean only cleans files built by make. Your problem is that CMakeCache.txt needed to be regenerated, and that is created by cmake not make.