f4exb/sdrangel #1521: MacOS 13.0.1 (ventura) - HackRF - Could not start sample source

moasson, 2022-11-22:
Hi,

As the title states, I have a macOS Ventura (13.0.1) with HackRF connected via a USB. SDRangel installs just fine, but when I click purple "Play" button, I get a screen pop up that says "Could not start sample source". That is all that it shows. Any help would be appreciated. 

[image]

srcejon, 2022-11-22:
I don't have an HackRF to try. You might try looking in the log file (Perferences > Logging menu) to see if that gives you any clues.

moasson, 2022-11-22:
Here is the result of the DEBUG mode from the sdrangel.log file

```
2022-11-22 15:14:02.605 (D) HackRFInput::handleMessage: MsgStartStop:  start
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::initAcquisition
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionInit
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoIdle
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoInit:   m_deviceDescription:  HackRF  sampleRate:  2400000  centerFrequency:  435000000
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoInit: initializing  SpectrumVis
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::startAcquisition
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStart
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoRunning
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoRunning:  HackRF  started
2022-11-22 15:14:02.605 (D) DSPDeviceSourceEngine::gotoError:  "Could not start sample source"
2022-11-22 15:14:02.607 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  435000000  sampleRate:  2400000
2022-11-22 15:14:02.607 (D) HackRFGui::handleInputMessages: message: DSPSignalNotification
2022-11-22 15:14:02.607 (D) HackRFGui::handleInputMessages: DSPSignalNotification: SampleRate:2400000, CenterFrequency:435000000
2022-11-22 15:14:02.607 (D) GLSpectrumGUI::handleInputMessages: message: GLSpectrumView::MsgReportSampleRate
2022-11-22 15:14:02.902 (D) DSPDeviceSourceEngine::errorMessage
2022-11-22 15:14:02.902 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPGetErrorMessage
2022-11-22 15:14:06.841 (D) HackRFInput::handleMessage: MsgStartStop:  stop
2022-11-22 15:14:06.841 (D) DSPDeviceSourceEngine::stopAcquistion
2022-11-22 15:14:06.841 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2022-11-22 15:14:06.842 (D) DSPDeviceSourceEngine::gotoIdle
```

closed 2023-01-27 by github-actions[bot]