f4exb/sdrangel #1524: Automatic location retrieving over GPS (serial)

bddvlpr, 2022-11-27:
It'd be a great addition to automatically set the user's location using (for example) [NMEA 0183](https://en.wikipedia.org/wiki/NMEA_0183) over serial port / usb. This would help out a lot if the user is on the move (in a car, train or any vehicle).

srcejon, 2022-11-27:
Funnily enough, I just wrote the code for this the other day.   It will be added with a few other things I'm working on.

bddvlpr, 2022-11-27:
Great, thanks a lot!

f4exb, 2022-11-28:
There are two ways of implementing this: in the main application or as a feature plugin. Maybe a feature plugin would be more flexible and hence desirable.

srcejon, 2022-12-06:
> There are two ways of implementing this: in the main application or as a feature plugin. Maybe a feature plugin would be more flexible and hence desirable.

At the moment, it's in MainCore, as it's only ~15 lines of code, so a plugin seemed like overkill (it's just using a Qt API with no user-configuration), and it's beneficial to start-up the GPS as soon as possible, as it can take a little while to get a valid position. But, of course, can be moved if desired.

f4exb, 2022-12-07:
It depends on how this is implemented. How does it deal with the serial interface? Does it do some sort of scanning? Will it delay program startup?

Also I was wondering if you would like to implement some fancy display like satellites positions (I' m hoping to find the right argument here :smile:  ) which would be easier to implement in an independent plugin. However I don' t know if this is part of the NMEA messages sent by the GPS.

bddvlpr, 2022-12-07:
Most NMEA compliant devices send the satellites information like azimuth, elevation, id, etc. using the [GPGSV](https://docs.novatel.com/OEM7/Content/Logs/GPGSV.htm) message to up to four satellites in one message.

srcejon, 2022-12-07:
Yeah. It was from me. Should be safe!

f4exb, 2022-12-07:
For Android... Just amazing! I gave up on this a long time ago: https://github.com/f4exb/sdrangel/issues/125

For GPS, yes that looks good like this we can see later if we need anything more fancy. Of course on an Android device with GPS (is there any without?) it is totally seamless!

srcejon, 2023-01-17:
This should be in 7.8.6.

Try setting the environment variable QT_NMEA_SERIAL_PORT to the Com port / serial device, before starting SDRangel.

closed 2023-02-13 by srcejon