f4exb/sdrangel #1530: Request for adding FLEX-1600 in the Pager Demodulator

jaapcammeraat, 2022-12-10:


jaapcammeraat, 2022-12-10:
This software has FLEX-1600 implemented and used by a lot of other developers.
https://github.com/Zanoroy/multimon-ng

Used by emergency services in Holland:
https://nl.wikipedia.org/wiki/P2000_(netwerk)

Example of this kind of data:
https://expert.p2kflex.nl

flipvkg, 2023-06-04:
Also used in Australia by various agencies and private entities,
Believe I did have a feature request on this, would be a good add-on, as am having issues with PDW of late

robbietee, 2023-07-20:
Ability to decode FLEX would be a great improvement. I'm quite surprised it hasn't been done already with all the other great features. Maybe nobody is interested in pagers  anymore?

rdwilson, 2023-12-24:
Spok uses FLEX-6400 in most of the USA. I would welcome this feature as well.

keukenrol, 2024-01-15:
There are multiple alternative softwares, but an implementation in SDRAngel would be a plus certainly.

CyborgArmy83, 2024-04-24:
Would also love FLEX-1600 support in SDRAngel! Is there a way it can be done with custom modules or is that not supported? I am not on Windows so I can't just use PDW or similar program.

temmink, 2025-01-12:
indeed, Flex-1600 support would be great!.

srcejon, 2026-10-27:
Implemented in v8.

closed 2026-10-27 by srcejon