f4exb/sdrangel #1533: Waterfall screen recording

AMIN2029, 2022-12-14:
I am looking for a way to record waterfall spectrogram screen   which update N frames per second.

I plan to capture and save waterfall screen  images(or video) and use them for further processing.

If there is a way to do this now I am not finding it?

I have looked at the code and I am thinking this "glspectrumgui"  cpp file will be  changed for recording waterfall screen.

Do you have an idea which function in the "glspectrumgui" file I can use  to record waterfall screen?

If this feature already exists in some way, my apologies in advance.
I am grateful for your help.

srcejon, 2022-12-14:
Glspectrumgui is just the widgets underneath – glspectrumview has the actual spectrum.

Might be easier to just use some desktop recording s/w.

AMIN2029, 2022-12-14:
> Glspectrumgui is just the widgets underneath – glspectrumview has the actual spectrum.
> 
> Might be easier to just use some desktop recording s/w.

[image]

thanks a lot
 
  SDRAngel is the best available platform for SDR development, and I plan to add a feature to SDRAngel that receives waterfall images and uses deep learning to detect and classify RF signals, as shown in the image above. (Wifi and Bluetooth channels have been detected)

  In the first stage, it works based on image processing and the shape of the occupied channel in the waterfall spectrum, and in the next stage, based on the analysis of I-Q signals and using the Radio Frequency Machine Learning (RFML) project.

Desktop recorder tools are not suitable for this feature, especially when user opens another workspace in SDRAngel .
For this reason, I need direct access to the waterfall screen.
thanks for your help .

srcejon, 2022-12-14:
The displayed spectrum is just the magnitude of the FFT, converted to colour through a lookup table - so rather than hooking in into glspectrumview, you might want to consider add a plugin feature or channel, that works on the FFT data itself.

Alternatively, you may be able to get the spectrum image via websockets: https://github.com/f4exb/sdrangel/issues/224

closed 2022-12-14 by AMIN2029