f4exb/sdrangel #1536: Install Sdrangel via Qt6

choupette-change, 2022-12-17:
Hello,
After installing Qt6 dependencies, installing Sdrangel crash, can you help me?? (Ubuntu 22 04)
tutu@tutu-:~$ cd /opt/build/sdrangel
tutu@tutu-:/opt/build/sdrangel$ cmake --preset default-qt6
Preset CMake variables:

  AIRSPYHF_DIR="/opt/install/libairspyhf"
  AIRSPY_DIR="/opt/install/libairspy"
  APT_DIR="/opt/install/aptdec"
  BLADERF_DIR="/opt/install/libbladeRF"
  CM256CC_DIR="/opt/install/cm256cc"
  CMAKE_INSTALL_PREFIX="/opt/install/sdrangel"
  CODEC2_DIR="/opt/install/codec2"
  DAB_DIR="/opt/install/libdab"
  DEBUG_OUTPUT="ON"
  DSDCC_DIR="/opt/install/dsdcc"
  ENABLE_QT6="ON"
  HACKRF_DIR="/opt/install/libhackrf"
  IIO_DIR="/opt/install/libiio"
  LIBSIGMF_DIR="/opt/install/libsigmf"
  LIMESUITE_DIR="/opt/install/LimeSuite"
  MBE_DIR="/opt/install/mbelib"
  MIRISDR_DIR="/opt/install/libmirisdr"
  PERSEUS_DIR="/opt/install/libperseus"
  RTLSDR_DIR="/opt/install/librtlsdr"
  SERIALDV_DIR="/opt/install/serialdv"
  SGP4_DIR="/opt/install/sgp4"
  SOAPYSDR_DIR="/opt/install/SoapySDR"
  UHD_DIR="/opt/install/uhd"
  XTRX_DIR="/opt/install/xtrx-images"

-- SDRangel Version: 7.8.5
-- Build type set to: Release
-- Compiling for 24 bit Rx DSP chain
-- Target architecture: x86_64-native
-- Looking for __AVX__ - forced found
-- Looking for __SSE4_2__ - forced found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
-- Interprocedural optimization enabled
-- Could NOT find Qt6Multimedia (missing: Qt6Multimedia_DIR)
-- Could NOT find Qt6MultimediaWidgets (missing: Qt6MultimediaWidgets_DIR)
-- Could NOT find Qt6Positioning (missing: Qt6Positioning_DIR)
CMake Warning at CMakeLists.txt:417 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt component "Multimedia".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6Multimedia/Qt6MultimediaConfig.cmake"
  does NOT exist

  Failed to find Qt component "MultimediaWidgets".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6MultimediaWidgets/Qt6MultimediaWidgetsConfig.cmake"
  does NOT exist

  Failed to find Qt component "Positioning".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6Positioning/Qt6PositioningConfig.cmake"
  does NOT exist

  

-- Could NOT find Qt6Quick (missing: Qt6Quick_DIR)
-- Could NOT find Qt6QuickWidgets (missing: Qt6QuickWidgets_DIR)
-- Could NOT find Qt6WebEngineQuick (missing: Qt6WebEngineQuick_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake:269 (message):
  Failed to find Qt component "WebEngineQuick".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6WebEngineQuick/Qt6WebEngineQuickConfig.cmake"
  does NOT exist

Call Stack (most recent call first):
  CMakeLists.txt:445 (find_package)

-- Could NOT find Qt6WebEngineCore (missing: Qt6WebEngineCore_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake:269 (message):
  Failed to find Qt component "WebEngineCore".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6WebEngineCore/Qt6WebEngineCoreConfig.cmake"
  does NOT exist

Call Stack (most recent call first):
  CMakeLists.txt:445 (find_package)

-- Could NOT find Qt6WebEngineWidgets (missing: Qt6WebEngineWidgets_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake:269 (message):
  Failed to find Qt component "WebEngineWidgets".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6WebEngineWidgets/Qt6WebEngineWidgetsConfig.cmake"
  does NOT exist

Call Stack (most recent call first):
  CMakeLists.txt:445 (find_package)

-- Could NOT find Qt6TextToSpeech (missing: Qt6TextToSpeech_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake:269 (message):
  Failed to find Qt component "TextToSpeech".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6TextToSpeech/Qt6TextToSpeechConfig.cmake"
  does NOT exist

Call Stack (most recent call first):
  CMakeLists.txt:445 (find_package)

CMake Error at CMakeLists.txt:445 (find_package):
  Found package configuration file:

    /usr/lib/x86_64-linux-gnu/cmake/Qt6/Qt6Config.cmake

  but it set Qt6_FOUND to FALSE so package "Qt6" is considered to be NOT
  FOUND.  Reason given by package:

  Failed to find Qt component "Quick".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6Quick/Qt6QuickConfig.cmake" does NOT
  exist

  Failed to find Qt component "QuickWidgets".

  Expected Config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt6QuickWidgets/Qt6QuickWidgetsConfig.cmake"
  does NOT exist

  

-- Configuring incomplete, errors occurred!
See also "/opt/build/sdrangel/build-qt6/CMakeFiles/CMakeOutput.log".
See also "/opt/build/sdrangel/build-qt6/CMakeFiles/CMakeError.log".
tutu@tutu-Vivobook-ASUSLaptop-X1702ZA-X1702ZA:/opt/build/sdrangel$

srcejon, 2022-12-17:
Please stick to qt5. Qt6 isn't ready yet.

f4exb, 2022-12-17:
Yes... this can be closed

closed 2022-12-17 by f4exb