f4exb/sdrangel #1537: 7.8.4 and 7.8.5 either fails to open or takes 5 Minutes when W10 is update to 22H2.

mistergitj, 2022-12-17:
I tried 7.8.5 on my W10 system at Version 22H2. 7.8.4 takes at least 5 Minutes to load. 7.8.4 exhibits this same behavior, but runs fine on W10 22H1. Thanks, John.

srcejon, 2022-12-28:
Just updated Windows, and no change in startup time for me.

Please upload the log file, which will have timestamps for each individual step (Preferences > Logging menu)

mistergitj, 2022-12-28:
Thanks, srcejon. Just started another test on 7.8.5; 4 and a half minutes gone, came up about 30 seconds ago but refuses to log. Please see my SDRangel [thread](https://groups.io/g/sdrangel/topic/w10_and_v7_8_3_really_long/95650233?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,0,95650233,previd%3D1671849024642443166,nextid%3D1668860017020030774&previd=1671849024642443166&nextid=1668860017020030774).
What version of W10 (winver command in Command Prompt) are your running? I am on W10 22H219045.2364. Thanks and enjoy, John.

srcejon, 2022-12-29:
Please send a screenshot showing the logging preferences window.

mistergitj, 2022-12-29:
[image]

srcejon, 2022-12-29:
Is a file created at all? i.e. is it just an empty file, or no file there at all?

mistergitj, 2022-12-29:
Thanks, srcejon. initially I let the Log file default to the SDRangel folder but was told users could not write to it. Worked fine in the past but I changed to my Desktop. In neither folder was a file created.
As a curiosity some time ago I looked at the QT site. I know nothing about QT and am not a programmer. But it looks like to me QT 6.5 is required for W10 22H2 and QT 6.4 for W11 22H2. Enjoy, John.

srcejon, 2023-01-03:
It's starting fine for me on both W10 22H2 and W11 22H2.

Perhaps trying unplugging your SDRs (which do you have?) and see if that makes a difference to the startup time.

mistergitj, 2023-01-03:
Thanks, srcejon. I just started SDRangel on my W10 22H2 with no SDR attached and it took 5 minutes. There is also no log. I usually run AirSpy R2. Thanks, John.

srcejon, 2023-01-04:
Try installing this test build:

http://sdrangel.org/builds/sdrangel-7.8.5-596-g0b476f6dd-win64.exe

When you run SDRangel, it should open a console window and write the log to that. Try to get a screenshot that shows where the longest delay is.

mistergitj, 2023-01-04:
Thanks, srcejon. Will do but was heading to lunch when I saw your post. Thanks and enjoy, John.

mistergitj, 2023-01-04:
srcejon, I got four screenshots and compressed them in a folder.  Two of them have a white space at the top which I will investigate and send you new ones if you like but I wanted to get these to you ASAP. Enjoy, John.
[SDRangelLog.zip](https://github.com/f4exb/sdrangel/files/10346291/SDRangelLog.zip)

srcejon, 2023-01-05:
The log seems to show it only taking a few seconds. At what point is the pause which results it taking 5 minutes? Before the log window appears? After all that is printed?

mistergitj, 2023-01-05:
Right, screjon, I notice that the log showed nothing (at least to me) interesting and took less than two minutes. I will repeat it this morning and watch the real time. What I see is nothing after starting SDRangel till about 5 minutes then it comes up. Thanks and enjoy, John.

srcejon, 2023-01-05:
>I see is nothing after starting SDRangel till about 5 minutes then it comes up. 

Until what comes up? The log console, the splash screen or the main window?

If nothing appears for five minutes, try switching off any virus checkers, and the like.

mistergitj, 2023-01-05:
Should have said two seconds not 2 minutes. Once this became a repeatable problem, I quit watching. I will watch this morning. Usually when my AV is a burden I can see it in the task manager. At first it looked like I never saw the splash but later noticed it did come up, maybe just before main. Above I talked about QT versions but never got a response. Is this relevant at all? Could you build a test bin with QT6.5? Will start the test now. Thanks, John.

srcejon, 2023-01-05:
Qt 6 isn't relevant.

srcejon, 2023-01-05:
>Command prompt opens quickly after starting, at about 5 minutes logging begins in command prompt

This build includes some additional debug between when the console opening and the first message in your log:

http://sdrangel.org/builds/sdrangel-7.8.5-598-g20bdbfa30-win64.exe

Please post a screenshot of what it outputs, thanks.

srcejon, 2023-01-05:
Well, the pause corresponds to the following single line of code in app/main.cpp

	QApplication a(argc, argv);

No idea why that would take 5 minutes.

mistergitj, 2023-01-05:
Sorry, srcejon, I am not a programmer, so means little to me. Glad you got it narrowed down. Enjoy, John.

mistergitj, 2023-01-14:
Thanks for 7.8.6 but it also takes 5 minutes to open in W10 22H2. Enjoy, John.

mistergitj, 2023-01-25:
I posted this in the wrong thread about startup issues:

Weird! I started SDRangel 7.8.6 today and it came up immediately (no 5 minute lag) and logging is working. I do not have a clue what happened except there was a W10 update a week ago. Enjoy, John.

This was posted several days ago, but all continues well today. I assume it was a W11 update. Thanks and enjoy, John.

closed 2023-03-26 by github-actions[bot]