f4exb/sdrangel #1547: Failure to launch?

SilentCyandFriends, 2022-12-25:
I'm running Linux Mint and am relatively new to the OS, so I installed it using flatpak, but it won't open? Any help would be appreciated!

srcejon, 2022-12-27:
Is there an error message if you run it from the command line?

jargowares, 2023-01-25:
It core dumps. Latest version of Linux Mint as of Jan 25th 2023.
joel@joel-Inspiron-3521:/var/lib/systemd/coredump$ ls -Flart
total 4880
drwxr-xr-x  13 root root   4096 Dec  1 22:02 ../
-rw-r-----+  1 root root 560518 Jan 24 23:12 core.sdrangel.1000.b68190f0c6df44d3845241ea7c0cd6f5.3475.1674623578000000.zst
-rw-r-----+  1 root root 560270 Jan 24 23:50 core.sdrangel.1000.6c15eaf66b3f493db3a2c92e30638460.3786.1674625835000000.zst
-rw-r-----+  1 root root 560142 Jan 24 23:50 core.sdrangel.1000.6c15eaf66b3f493db3a2c92e30638460.3804.1674625847000000.zst
-rw-r-----+  1 root root 560827 Jan 24 23:54 core.sdrangel.1000.6c15eaf66b3f493db3a2c92e30638460.3946.1674626078000000.zst
-rw-r-----+  1 root root 457437 Jan 25 01:48 core.sdrangel.1000.b6284e69fa08486cacb7c9410f4d9d8e.4413.1674632919000000.zst
-rw-r-----+  1 root root 560398 Jan 25 01:54 core.sdrangel.1000.b6284e69fa08486cacb7c9410f4d9d8e.5859.1674633284000000.zst
-rw-r-----+  1 root root 561254 Jan 25 02:02 core.sdrangel.1000.b6284e69fa08486cacb7c9410f4d9d8e.6486.1674633746000000.zst
-rw-r-----+  1 root root 560733 Jan 25 03:07 core.sdrangel.1000.7e6bd4a3334140998acbdceae4c13757.2494.1674637634000000.zst
-rw-r-----+  1 root root 560695 Jan 25 03:34 core.sdrangel.1000.7e6bd4a3334140998acbdceae4c13757.6074.1674639259000000.zst
drwxr-xr-x   2 root root   4096 Jan 25 03:34 ./
joel@joel-Inspiron-3521:/var/lib/systemd/coredump$

srcejon, 2023-01-25:
Can you start it under gdb?

    gdb /path/to/sdrangel
    run

Then when it crashes, type:

    info stack

And copy and paste the output here?

jargowares, 2023-01-26:
(gdb) file sdrangel
Reading symbols from sdrangel...
Reading symbols from /usr/lib/debug//app/bin/sdrangel.debug...
(gdb) run
Starting program: /app/bin/sdrangel
[Thread debugging using libthread_db enabled]
Using host libthread_db library 
"/usr/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00007ffff7588f6c in 
std::__detail::_Compiler<std::__cxx11::regex_traits<char> 
 >::_Compiler(char const*, char const*, std::locale const&, 
std::regex_constants::syntax_option_type) [clone .constprop.0] 
(this=0x7fffffffd9f0, __b=<optimized out>, __e=<optimized out>, 
__loc=..., __flags=std::regex_constants::_S_ECMAScript) at 
/usr/include/c++/12.1.0/bits/regex_compiler.tcc:72
72          _M_ctype(std::use_facet<_CtypeT>(__loc))
(gdb) info stack
#0  0x00007ffff7588f6c in 
std::__detail::_Compiler<std::__cxx11::regex_traits<char> 
 >::_Compiler(char const*, char const*, std::locale const&, 
std::regex_constants::syntax_option_type) [clone .constprop.0] 
(this=0x7fffffffd9f0, __b=<optimized out>,
     __e=<optimized out>, __loc=..., 
__flags=std::regex_constants::_S_ECMAScript)
     at /usr/include/c++/12.1.0/bits/regex_compiler.tcc:72
#1  0x00007ffff743ce50 in std::__cxx11::basic_regex<char, 
std::__cxx11::regex_traits<char> >::_M_compile(char const*, char const*, 
std::regex_constants::syntax_option_type) [clone .constprop.0] (
     ***@***.***=0x7ffff7cfed40 
<WebAPIAdapterInterface::devicesetURLRe[abi:cxx11]>,
     ***@***.***=0x7ffff7c9ae08 
"^/sdrangel/deviceset/([0-9]{1,2})$", __last=<optimized out>,
     __f=std::regex_constants::_S_ECMAScript) at 
/usr/include/c++/12.1.0/bits/regex.h:789
#2  0x00007ffff743d023 in std::__cxx11::basic_regex<char, 
std::__cxx11::regex_traits<char> >::basic_regex(char const*, 
std::regex_constants::syntax_option_type) [clone .constprop.0] 
(this=0x7ffff7cfed40 <WebAPIAdapterInterface::devicesetURLRe[abi:cxx11]>,
     __p=0x7ffff7c9ae08 "^/sdrangel/deviceset/([0-9]{1,2})$", 
__f=std::regex_constants::_S_ECMAScript)
     at /usr/include/c++/12.1.0/bits/regex.h:453
#3  0x00007ffff7442858 in _sub_I_65535_0.0 () from 
/app/lib/sdrangel/libsdrbase.so
#4  0x00007ffff7fcbfff in call_init () from /lib64/ld-linux-x86-64.so.2
#5  0x00007ffff7fcc0fd in _dl_init () from /lib64/ld-linux-x86-64.so.2
#6  0x00007ffff7fe41ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#7  0x0000000000000001 in ?? ()
#8  0x00007fffffffe0a6 in ?? ()
#9  0x0000000000000000 in ?? ()
(gdb)

srcejon, 2023-01-26:
Problem is "illegal instruction". Do you have an old CPU? If so, you may need to recompile for your CPU.

jargowares, 2023-01-27:
Is this too old?

Intel(R) Core(TM) i3-3217U CPU @ 1.80GHz

Joel

srcejon, 2023-01-29:
I don't know what options the flatpak was compiled with, but as that CPU looks to be 10 years old, it may be.

I'd suggest trying to compile SDRangel yourself.

https://github.com/f4exb/sdrangel/wiki/Compile-from-source-in-Linux

closed 2023-03-25 by github-actions[bot]