f4exb/sdrangel #1548: sdrangel startup issue

nico-pret, 2022-12-27:
Hi 
My problem is with SDRANGEL on windows11 .it was working and all the sudden stopped working.when I opens then it opens with the menu on top and a message on the bottom.How do i get out of this mode?.I tried to uninstall and install again...but same problem.why is this happening ?
it was working perfectly on windows11.can you advise...might it be one of the drivers on windows 11 that updated ..and now it crashed?

[image]

srcejon, 2022-12-27:
What happens if you do workspace > new?

mistergitj, 2022-12-28:
nico-pret, what version (winver command in Command Prompt) of W11 are you running? W10 22H2 causes very long startup for me, see [here](https://github.com/f4exb/sdrangel/issues/1537). Enjoy, John.

mistergitj, 2023-01-21:
Weird! I started SDRangel 7.8.6 today and it came up immediately (no 15 minute lag) and logging is working. I do not have a clue what happened except there was a W10 update a week ago. Enjoy, John.

TennTrips, 2023-03-05:
Same issue.  Pull down menu's don't work and the only option is to "end task" in task manager.
Worked GREAT for 5 minutes after install, started looking around and everything disappeared and menu's quit working.  Restarted program, no help.  Restarted computer, no help.  Uninstalled and re-installed, NO HELP...

HELP ! ! !

ThermoNuclear65, 2023-03-27:
I un-docked the workspace , pulled it down a bit , dropdowns and tooltips working again.

ThermoNuclear65, 2023-03-27:
Second trick , change resolution to a ridicule low setting until you see the windows frame with "SDR Angel" written in it then the delete the empty workspace .

closed 2023-05-12 by github-actions[bot]