f4exb/sdrangel #1553: Download 7.8.5 results in 7.8.4 installed

mistergitj, 2022-12-31:


srcejon, 2023-01-03:
I see version 7.8.5 reported in the status bar.

mistergitj, 2023-01-03:
Thanks, screjon. Something really strange is going on here. On my 22H2 system I am running:
[image]
Notice that it does not have the extra numbers after the 7.8.5 in the name. I just installed using the DL labeled 7.8.5 on my 21H2 system and got 7.8.4-6-g92... This was DLed a couple of days ago. So, two 7.8.5 installers DLed on different days install different version. Thanks and enjoy, John.

mistergitj, 2023-01-03:
screjon, I suspect you think I am a nut, but I had a similar problem with Comodo some time ago. Here is a link to the [thread](https://forums.comodo.com/news-announcements-feedback-cis/comodo-internet-security-2019-v12006818-ndash-released-t124172.210.html). If you will search on johnhc you will begin to see my posts. They did not finally admit to a problem until the next page but it turned out to be a regional problem affecting some but not all. Thanks and enjoy, John.

mistergitj, 2023-01-21:
I took a half day to chase this problem down and found my AV (Comodo) had contained SDRangel by placing it in the VTroot folder. When I installed a new version it asked if I wanted to uninstall the current and I always answer yes and the install proceeded. Actually the old version was not removed and the new version was installed in the C:\VTroot folder. I know using Windows was a problem but SDRangel should have told me the uninstall failed and I would have looked into the problem. The failed uninstalled version stayed put and ran when SDRangel was executed. I still do not know why Comodo contained SDRangel. Enjoy, John.

closed 2023-03-25 by github-actions[bot]