f4exb/sdrangel #1557: LimeRFE RX mode not enabled after changing RX channel settings.

srcejon, 2023-01-05:
I'm trying to use the LimeRFE feature in RX mode. It seems after changing RX channel settings, I need to uncheck, then recheck the RX button in order to receive a signal.

To recreate:

- Setup lime RFE with HAM 144-146MHz RX channel selected.
- Enable RX
- RX button goes green and signal is received
- Change channel to 220-225MHz
- Press apply
- Change channel back to 144-146MHz
- Press apply
- No signal (or only a very weak signal) is received, by RX button remains green
- Disable RX
- Enable RX
- Signal is now received.

Log:

2023-01-05 16:14:54.580 (D) LimeRFE::configure:  attValue:  0 channelIDRX:  6 channelIDTX:  1 mode:  2 notchOnOff:  0 selPortRX:  1 selPortTX:  2 enableSWR:  0 sourceSWR:  0
2023-01-05 16:14:54.628 (D) LimeRFE::configure: done
2023-01-05 16:14:56.981 (D) LimeRFE::configure:  attValue:  0 channelIDRX:  5 channelIDTX:  1 mode:  2 notchOnOff:  0 selPortRX:  1 selPortTX:  2 enableSWR:  0 sourceSWR:  0
2023-01-05 16:14:57.039 (D) LimeRFE::configure: done
2023-01-05 16:14:57.997 (D) LimeRFE::setRx: switch off mode: 2
2023-01-05 16:14:58.253 (D) LimeRFE::setRx: switch on mode: 0

Macintoshsh, 2023-01-08:
i cant connect LimeSDR in SDRAngel.  lime sdr drivers is presented.

srcejon, 2023-01-08:
> i cant connect LimeSDR in SDRAngel. lime sdr drivers is presented.

Please open another issue including what OS you are using and more information such as a log file (Menu Preferences > Logging) or screenshots.

f4exb, 2023-01-14:
Be also aware that LimeSuite support for Windows was removed after v7.8.3 and re-instated just recently in v7.8.6.

f4exb, 2023-01-14:
I replayed the same sequence (I can switch filters forth and back) and reproduced the issue.

f4exb, 2023-01-14:
When the mode is switch to Rx, Tx or both the RFE mode is not saved in the `m_rfeBoardState.mode` variable therefore when changing the RFE settings later like changing band filter it comes back to default which is no Rx and no Tx.

commit 0a257f39 (2023-01-14): LimeRFE: save Rx/Tx mode in m_rfeBoardState. Fixes #1557
closed 2023-01-14 by f4exb