f4exb/sdrangel #1562: Radio clock TDF demodulation is not working?

kord1e, 2023-01-07:
I think the TDF radio clock demodulator may not be working. Can someone confirm? 

Or is there something wrong with the actual signal on 162 kHz? 

MSF and DCF77 demodulation work well.

f4exb, 2023-01-07:
it does work if you have good reception. You have to be a little bit patient so that it acquires the minute mark and starts decoding:
[image]

kord1e, 2023-01-21:
@f4exb
Ok, thanks for confirming, maybe signal level is the problem then. But TDF 162 kHz signal seems quite strong here and I think I can even see the minute marker coming, it should be visible in this picture I think... But decoding doesn't start. I've tried different bandwidth and threshold options too.

[image]

srcejon, 2023-02-09:
It's working OK for me too

[image]

In the scope, set the trace to FM (in red box) and also include the centre frequency (other red box) in the screenshot please.

kord1e, 2023-02-14:
I tuned the clock a bit lower than the top peak of the signal, set the scope to FM. Then I fine adjusted the frequency until the scope signal was centered. Now it finds the minute marker but loses it usually after 20 seconds or so.

I tried changing the bandwidth and threshold values but it didn't help, it still finds the minute marker but loses it before another marker comes.

Does this mean that the signal is too weak? The peak is almost 40 dB above noise floor.

[image]

srcejon, 2023-02-14:
What sdr are you using and what are your settings? Seems a bit odd the frequency is that far off (but i'd have to go double check)

f4exb, 2023-02-15:
From the device UI behind I can guess it is a RTL-SDR and that the "No-mod DS" (for direct sampling) is not checked. So I guess that at this very low frequency for RTL-SDR some converter sits on the front and its LO may not be very accurate.

closed 2023-04-03 by github-actions[bot]

Ronda-00, 2024-03-30:
please
FYI receiving antenna is a MLA-30+ with a double loop

What diameter is the double loop and separation?

Thanks