f4exb/sdrangel #1563: V7.8.6 Segfault 22.04

alphafox02, 2023-01-14:
Sorry for the vagueness. 

System setup 22.04 on x86_86

Previous SDRAngel deb loads/works no issues.

Latest v7.8.6 release deb installed.
Starting from terminal I see no mentions of errors prior to it completing or almost completing the Device Enumerator section before it just goes right to Segmentation fault (core dump). 

I’ll try another clean 22.04 system when I get time, but again this just started. Perhaps there’s a new dependency that’s needed?

alphafox02, 2023-01-14:
Not sure if this helps, but it's what valgrind spits out.

`2023-01-14 05:40:44.101 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesink.hackrf
==363611== Invalid read of size 4
==363611==    at 0x6A9DF74: pthread_mutex_lock@@GLIBC_2.2.5 (pthread_mutex_lock.c:80)
==363611==    by 0x40141A36: libusb_get_device_list (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==363611==    by 0x40FF90F2: hackrf_device_list (in /usr/local/lib/libhackrf.so.0.7.0)
==363611==    by 0x41003E7B: DeviceHackRF::enumOriginDevices(QString const&, QList<PluginInterface::OriginDevice>&) (in /usr/lib/sdrangel/libhackrfdevice.so)
==363611==    by 0x4C40ACCB: HackRFOutputPlugin::enumOriginDevices(QStringList&, QList<PluginInterface::OriginDevice>&) (in /usr/lib/sdrangel/plugins/liboutputhackrf.so)
==363611==    by 0x4C4C83E: DeviceEnumerator::enumerateDevices(QList<PluginAPI::SamplingDeviceRegistration>&, std::vector<DeviceEnumerator::DeviceEnumeration, std::allocator<DeviceEnumerator::DeviceEnumeration> >&, PluginInterface::SamplingDevice::StreamType) (in /usr/lib/sdrangel/libsdrbase.so)
==363611==    by 0x4CB8B77: PluginManager::loadPluginsFinal() (in /usr/lib/sdrangel/libsdrbase.so)
==363611==    by 0x4908CDE: MainWindow::MainWindow(qtwebapp::LoggerWithFile*, MainParser const&, QWidget*) (in /usr/lib/sdrangel/libsdrgui.so)
==363611==    by 0x10B1E4: ??? (in /usr/bin/sdrangel)
==363611==    by 0x10A76A: main (in /usr/bin/sdrangel)
==363611==  Address 0x38 is not stack'd, malloc'd or (recently) free'd
==363611== 
==363611== 
==363611== Process terminating with default action of signal 11 (SIGSEGV)
==363611==  Access not within mapped region at address 0x38
==363611==    at 0x6A9DF74: pthread_mutex_lock@@GLIBC_2.2.5 (pthread_mutex_lock.c:80)
==363611==    by 0x40141A36: libusb_get_device_list (in /usr/lib/x86_64-linux-gnu/libusb-1.0.so.0.3.0)
==363611==    by 0x40FF90F2: hackrf_device_list (in /usr/local/lib/libhackrf.so.0.7.0)
==363611==    by 0x41003E7B: DeviceHackRF::enumOriginDevices(QString const&, QList<PluginInterface::OriginDevice>&) (in /usr/lib/sdrangel/libhackrfdevice.so)
==363611==    by 0x4C40ACCB: HackRFOutputPlugin::enumOriginDevices(QStringList&, QList<PluginInterface::OriginDevice>&) (in /usr/lib/sdrangel/plugins/liboutputhackrf.so)
==363611==    by 0x4C4C83E: DeviceEnumerator::enumerateDevices(QList<PluginAPI::SamplingDeviceRegistration>&, std::vector<DeviceEnumerator::DeviceEnumeration, std::allocator<DeviceEnumerator::DeviceEnumeration> >&, PluginInterface::SamplingDevice::StreamType) (in /usr/lib/sdrangel/libsdrbase.so)
==363611==    by 0x4CB8B77: PluginManager::loadPluginsFinal() (in /usr/lib/sdrangel/libsdrbase.so)
==363611==    by 0x4908CDE: MainWindow::MainWindow(qtwebapp::LoggerWithFile*, MainParser const&, QWidget*) (in /usr/lib/sdrangel/libsdrgui.so)
==363611==    by 0x10B1E4: ??? (in /usr/bin/sdrangel)
==363611==    by 0x10A76A: main (in /usr/bin/sdrangel)
==363611==  If you believe this happened as a result of a stack
==363611==  overflow in your program's main thread (unlikely but
==363611==  possible), you can try to increase the size of the
==363611==  main thread stack using the --main-stacksize= flag.
==363611==  The main thread stack size used in this run was 8388608.
==363611== 
==363611== HEAP SUMMARY:
==363611==     in use at exit: 8,053,358 bytes in 50,477 blocks
==363611==   total heap usage: 2,978,408 allocs, 2,927,931 frees, 286,569,191 bytes allocated
==363611== 
==363611== LEAK SUMMARY:
==363611==    definitely lost: 3,588 bytes in 40 blocks
==363611==    indirectly lost: 3,052 bytes in 80 blocks
==363611==      possibly lost: 176,300 bytes in 2,602 blocks
==363611==    still reachable: 7,868,402 bytes in 47,734 blocks
==363611==         suppressed: 0 bytes in 0 blocks
==363611== Rerun with --leak-check=full to see details of leaked memory
==363611== 
==363611== For lists of detected and suppressed errors, rerun with: -s
==363611== ERROR SUMMARY: 3 errors from 3 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
`

alphafox02, 2023-01-14:
Just to  make sure the previous release was still okay, I went and installed the 7.8.5 deb package. No issues with that version.

srcejon, 2023-01-14:
Try ruuning without hackrf connected

alphafox02, 2023-01-14:
Hackrf wasn’t connected, actually no SDR was connected.

srcejon, 2023-01-14:
Ok. The other thing to try is gdb.

    gdb sdrangel
    run

When it crashes, then type:

    info stack

And paste the output here.

alphafox02, 2023-01-14:
Might take me to this evening but I’ll do that. I should’ve thought of that earlier, sorry.

alphafox02, 2023-01-14:
A friend changed a clean install of 22.04 and doesn’t have the issue, so I’ll check again tonight and see what’s gone wrong with my particular setup.

alphafox02, 2023-01-14:
I checked on another system myself and it’s fine. I’ll close this and paste what I find from the troubled system since it’s likely not an issue with SDRAngel itself.

closed 2023-01-14 by alphafox02

alphafox02, 2023-01-15:
On the system that's having an issue, here's where it's at even after sitting overnight with gdb run. It won't proceed further nor will it crash.

```
2023-01-14 22:18:17.830 (D) PluginManager::registerTxChannel  WFM Modulator  with channel name  "sdrangel.channeltx.modwfm"
2023-01-14 22:18:17.830 (D) DeviceEnumerator::enumerateDevices: sdrangel.samplesource.airspy
[New Thread 0x7fff4c73a640 (LWP 443484)]
[Thread 0x7fff737fe640 (LWP 443478) exited]
[Thread 0x7fff6bfff640 (LWP 443476) exited]
[Thread 0x7fff73fff640 (LWP 443477) exited]
[Thread 0x7fffa0cbf640 (LWP 443475) exited]
[Thread 0x7fffa14c0640 (LWP 443474) exited]
[Thread 0x7fffa1cc1640 (LWP 443473) exited]
[Thread 0x7fffa24c2640 (LWP 443472) exited]
[Thread 0x7fffa2cc3640 (LWP 443471) exited]
```

srcejon, 2023-01-15:
You could try deleting the airspy input plugin: /usr/lib/sdrangel/plugins/libinputairspy.so - see if that narrows it down.