f4exb/sdrangel #1567: No spectrum under Ubuntu 22.04.1 LTS

dwarning, 2023-01-16:
I installed latest .deb package on my old HP ProBook 6440b. I can plugin my SDR stick and my Airspy too and see that reception is working and can hear FM station very well. Problem is that I can't see a spectrum in the different plugins, e.g. Broadcast FM Demodulator. The small window is entire black. For me it looks like a graphic problem.
Made the same check under Win 10 everything is fine.
How can I check if my graphic capabilities fulfil the requirements? Is there something to configure?

srcejon, 2023-01-16:
What version of OpenGL is reported in the status bar at the bottom right of the window? Needs to be at least OpenGL 2.1. Perhaps post a screenshot.

 

You might also want to look in the log file (Preferences > Logging) to see if there are any relevant messages.

dwarning, 2023-01-16:
Thanks for the quick response.

This is my openGL version:

`client glx vendor string: Mesa Project and SGI
    Device: Mesa Intel(R) HD Graphics (ILK) (0x46)
OpenGL renderer string: Mesa Intel(R) HD Graphics (ILK)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 22.0.5
OpenGL version string: 4.5 (Compatibility Profile) Mesa 22.0.5
OpenGL ES profile version string: OpenGL ES 2.0 Mesa 22.0.5
`
And same OpenGL Version 4.5 is shown in the bottom message.

In the large logfile are error messages regarding GLSL library version 3.3:

`2023-01-16 12:50:46.518 (D) GLSpectrumView::initializeGL: context:  major:  4  minor:  5  ES:  no
2023-01-16 12:50:46.531 (W) QOpenGLShader::compile(Vertex): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

2023-01-16 12:50:46.531 (W) *** Problematic Vertex shader source code ***
#version 330
#define lowp
#define mediump
#define highp
#line 1
uniform highp mat4 uMatrix;
in highp vec4 vertex;
void main() {
    gl_Position = uMatrix * vertex;
}

***
2023-01-16 12:50:46.532 (D) GLShaderSimple::initializeGL: error in vertex shader:  "0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES\n"
2023-01-16 12:50:46.532 (W) QOpenGLShader::compile(Fragment): 0:1(10): error: GLSL 3.30 is not supported. Supported versions are: 1.10, 1.20, 1.30, 1.40, and 1.00 ES

2023-01-16 12:50:46.532 (W) *** Problematic Fragment shader source code ***
#version 330
#ifdef GL_KHR_blend_equation_advanced
#extension GL_ARB_fragment_coord_conventions : enable
#extension GL_KHR_blend_equation_advanced : enable
#endif
#define lowp
#define mediump
#define highp
#line 1
out vec4 fragColor;
uniform mediump vec4 uColour;
void main() {
    fragColor = uColour;
}
`
This sounds stupid, because of the requested versions 1.00 ... 1.40.

srcejon, 2023-01-16:
> This helps - but can't overlook the implications:
>
> MESA_GL_VERSION_OVERRIDE=3.3 MESA_GLSL_VERSION_OVERRIDE=330 sdrangel

The SDRangel code is checking the OpenGL version, and using that to decide what GLSL shader code to use. There are different shaders for >= 3.3 and < 3.3. 

 

However, it seems although you have OpenGL 4.5, your card doesn’t support GLSL 3.3 shaders. So, we probably need to check the GLSL version rather than the OpenGL version.

 

To get the older shader code, I’d guess you should try MESA_GL_VERSION_OVERRIDE=2.1 sdrangel

 

Cheers,

Jon

dwarning, 2023-01-16:
Perhaps a misunderstanding:
Initial OpenGL 4.5 didn't work.
My overwrite version to 3.3 works!
Your suggestion overwrite to 2.1 bring a total wrong application on the desktop - not usable.
log on terminal
`2023-01-16 14:49:51.263 (W) QOpenGLShader::link: error: linking with uncompiled/unspecialized shadererror: linking with uncompiled/unspecialized shader
`

closed 2023-03-24 by github-actions[bot]