f4exb/sdrangel #1572: Deleted channel in deviceSet causes crash on exit

srcejon, 2023-01-25:
To recreate

- Add workspace with RX device (can be Test Source)
- Add 2 VOR Demodulators (not a problem with only 1)
- Add VOR Localizer
- Start device
- Start VOR Localizer
- Exit SDRangel (without pressing stop)
- It should crash - seems to be different on Linux and Windows

On Linux:

2023-01-25 11:28:15.530 (D) ChannelGUI::~ChannelGUI
2023-01-25 11:28:15.531 (D) ChannelGUI::~ChannelGUI: end
2023-01-25 11:28:15.531 (D) VORDemod::~VORDemod
2023-01-25 11:28:15.531 (D) DSPDeviceSourceEngine::removeSink:  VORDemod

Thread 27 "DSPDeviceSource" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff5ae8e640 (LWP 163126)]
0x00007ffff6107dac in QCoreApplication::notifyInternal2(QObject*, QEvent*) () from /lib/x86_64-linux-gnu/libQt5Core.so.5
(gdb) info stack
#0  0x00007ffff6107dac in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#1  0x00007ffff61603eb in QTimerInfoList::activateTimers() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007ffff6160cec in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007ffff43cbd1b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff44206f8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff43c93c3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff61610b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#7  0x00007ffff610675b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007ffff5f19af2 in QThread::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007ffff72a4cfb in DSPDeviceSourceEngine::run() () at /opt/install/sdrangel/lib/sdrangel/libsdrbase.so
#10 0x00007ffff5f1aca1 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff5a70b43 in start_thread (arg=<optimised out>) at ./nptl/pthread_create.c:442
#12 0x00007ffff5b02a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) 

Which isn't overly helpful.

On a Windows debug build:

2023-01-25 11:41:46.489 (D) MainWindow::closeEvent
2023-01-25 11:41:46.490 (D) MainWindow::saveConfiguration: configuration [default | no name] 2 workspaces
2023-01-25 11:41:46.492 (D) DeviceUISet::saveDeviceSetSettings: preset: [default, no name]
2023-01-25 11:41:46.493 (D) DeviceUISet::saveRxChannelSettings: saving channel [sdrangel.channel.vordemod]
2023-01-25 11:41:46.493 (D) DeviceUISet::saveRxChannelSettings: saving channel [sdrangel.channel.vordemod]
2023-01-25 11:41:46.495 (D) DeviceAPI::saveSamplingDeviceSettings: serializing source sdrangel.samplesource.fileinput[0]:
2023-01-25 11:41:46.495 (D) MainWindow::saveConfiguration: sdrangel.samplesource.fileinput device in workspace 1 spectrum in 1
2023-01-25 11:41:46.496 (D) FeatureUISet::saveFeatureSetSettings: saving feature [sdrangel.feature.vorlocalizer]
2023-01-25 11:41:46.497 (D) AudioDeviceManager::serialize
2023-01-25 11:41:46.497 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "CABLE Output (VB-Audio Virtual Cable)"  sampleRate:  48000  volume:  1
2023-01-25 11:41:46.498 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "Microphone (C922 Pro Stream Webcam)"  sampleRate:  48000  volume:  1
2023-01-25 11:41:46.498 (D) AudioDeviceManager::debugAudioInputInfos:  name:  "System default device"  sampleRate:  48000  volume:  1
2023-01-25 11:41:46.499 (D) AudioDeviceManager::debugAudioOutputInfos:  name:  "System default device"  sampleRate:  48000  udpAddress:  "127.0.0.1"  udpPort:  9998  copyToUDP:  false  udpUseRTP:  false  udpChannelMode:  0  udpChannelCodec:  0  decimationFactor:  1
2023-01-25 11:41:46.531 (D) DSPDeviceSourceEngine::stopAcquistion
2023-01-25 11:41:46.532 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2023-01-25 11:41:46.534 (D) DSPDeviceSourceEngine::gotoIdle
2023-01-25 11:41:46.534 (D) FileInput::stop
2023-01-25 11:41:46.534 (D) FileInputThread::stopWork
2023-01-25 11:41:46.538 (D) GLSpectrumGUI::handleInputMessages: message: SpectrumVis::MsgStartStop
2023-01-25 11:41:46.539 (D) VORDemod::stop
2023-01-25 11:41:47.160 (D) AudioDeviceManager::removeAudioSink: 0x10e23b6b230
2023-01-25 11:41:47.163 (D) VORDemod::stop
2023-01-25 11:41:47.247 (D) AudioDeviceManager::removeAudioSink: 0x10e1fe1f690
2023-01-25 11:41:47.252 (D) DSPDeviceSourceEngine::removeSink:  SpectrumVis
2023-01-25 11:41:47.253 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2023-01-25 11:41:47.254 (D) DeviceUISet::freeChannels: destroying channel [sdrangel.channel.vordemod]
2023-01-25 11:41:47.254 (D) ChannelGUI::~ChannelGUI
2023-01-25 11:41:47.256 (D) ChannelGUI::~ChannelGUI: end
2023-01-25 11:41:47.260 (D) VORDemod::~VORDemod
2023-01-25 11:41:47.260 (D) DSPDeviceSourceEngine::removeSink:  VORDemod
2023-01-25 11:41:47.261 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2023-01-25 11:41:47.262 (D) ObjectPipesRegistrations::removeProducer: 0x10e2375fe98 VORDemod
2023-01-25 11:41:47.263 (D) ObjectPipe::setToBeDeleted: 0 (0x10e2375fe98)
2023-01-25 11:41:47.263 (D) ObjectPipesRegistrations::registerProducerToConsumer: return existing pipe 0x10e2375fe98 0x10e2008cf90 MessageQueue report
2023-01-25 11:41:47.264 (D) ObjectPipesRegistrations::registerProducerToConsumer: return existing pipe 0x10e23760918 0x10e2008cf90 MessageQueue report
2023-01-25 11:41:47.265 (D) ObjectPipesRegistrations::registerProducerToConsumer: return existing pipe 0x10e2375fe98 0x10e2008cf90 MessageQueue report
2023-01-25 11:41:47.265 (D) VorLocalizerWorker::handleMessage: MsgRefreshChannels
2023-01-25 11:41:47.266 (D) ObjectPipesRegistrations::registerProducerToConsumer: return existing pipe 0x10e23760918 0x10e2008cf90 MessageQueue report
2023-01-25 11:41:47.268 (D) VorLocalizerWorker::updateChannels:  #VORs: 0 #Chans: 1
2023-01-25 11:41:47.270 (D) DeviceUISet::freeChannels: destroying channel [sdrangel.channel.vordemod]
2023-01-25 11:41:47.272 (D) VorLocalizerWorker::handleMessage: MsgRefreshChannels
2023-01-25 11:41:47.272 (D) ChannelGUI::~ChannelGUI
2023-01-25 11:41:47.273 (D) VorLocalizerWorker::updateChannels:  #VORs: 0 #Chans: 2
2023-01-25 11:41:47.274 (D) ChannelGUI::~ChannelGUI: end
2023-01-25 11:41:47.311 (D) VORDemod::~VORDemod
2023-01-25 11:41:47.311 (D) DSPDeviceSourceEngine::removeSink:  VORDemod
2023-01-25 11:41:47.314 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPRemoveBasebandSampleSink
2023-01-25 11:41:47.314 (D) ObjectPipesRegistrations::removeProducer: 0x10e23760918 VORDemod
2023-01-25 11:41:47.315 (D) ObjectPipe::setToBeDeleted: 0 (0x10e23760918)

[image]

Code in VORLocalizer::updateChannels()

            for (int chi = 0; chi < deviceSet->getNumberOfChannels(); chi++)
            {
                ChannelAPI *channel = deviceSet->getChannelAt(chi);
                if (channel->getURI() == "sdrangel.channel.vordemod")

channel is a pointer to an object that has been deleted.

Haven't looked further in to the code, but perhaps the channel needs to be removed from the deviceSet before it is deleted?

f4exb, 2023-01-27:
I think the call to `updateChannels` may simply be removed as we already removed the reference from the list of channels in the `handleMessagePipeToBeDeleted` method.

There are similar occurrences in the DempdAnalzyzer and JogDialController plugins.

f4exb, 2023-02-01:
I've tried the above but with no success. However in my case (Linux) this crash on exit is not systematic.

srcejon, 2023-02-02:
I think running with a debug build on windows helps to make the problem more reproducible, as it invalidates memory when delete is called - but with a release build, it's possible the freed memory doesn't always get overwritten, so wont necessarily crash.

f4exb, 2023-02-03:
In fact removing call to `updateChannels` in `handleMessagePipeToBeDeleted` does work. The second segfault was due to an issue with Test Source and timers. In fact one should connect worker's `deleteLater` with `Qt::QueuedConnection` else the destructor is called from the main thread and the timer cannot be stoped.
```
QObject::connect(m_testSourceWorkerThread, &QThread::finished, m_testSourceWorker, &QObject::deleteLater, Qt::QueuedConnection);
```

commit 46db55f1 (2023-02-03): Removed unncecessary call to updateChannels in VORLocalizer and DemodAnalyzer. Fixes #1572
closed 2023-02-03 by f4exb