f4exb/sdrangel #1573: Is DemodDSD packaged for Fedora?

Fwappy, 2023-01-26:
[The Quick Start Page](https://github.com/f4exb/sdrangel/wiki/Quick-start) only mentions Satellite Tracker, DAB Demod, and APT Demod being excluded, but I cannot find `libdemoddsd.so`

srcejon, 2023-01-26:
No, probably because of the dependency on mbelib. I've updated the Quick Start page.

f4exb, 2023-01-26:
Maybe mbelib dependency could be made optional. If you have an AMBE key in your system or access to an AMBE server you don't need mbelib. And if you don't have any of these you would not hear any sound but get the control messages. However the dependency tree is a bit complex. I am not even sure if the dependency is in SDRangel or the ancillary DSDcc project.

AsciiWolf, 2023-02-13:
It would be great if the mbelib dependency was optional. :-) I would like to not ship mbelib in my SDRangel Flatpak (and recommend users to use supported AMBE hardware dongle instead) because of possible legal issues, sadly I have to ship mbelib in order to be able to include the DSD decoder functionality in the Flatpak.

closed 2023-03-31 by github-actions[bot]