f4exb/sdrangel #1574: V7.8.6 The program did not respond and crashed when trying to open "AudioOutput".

minexixi, 2023-01-27:
V7.8.6 The program did not respond and crashed when trying to open "AudioOutput".
There is no ATV modulator in the mode list of the TX device.
There is no IEEE 802.15.4 demodulator in the mode list of RX devices.
The IEEE 802.15.4 modulator of TX device cannot open the window, and the main program does not respond.

srcejon, 2023-01-27:
What OS are you on.

There isn't an 802.15.4 demodulator. The modulator opens ok for me.

ATV missing is in https://github.com/f4exb/sdrangel/issues/1525

srcejon, 2023-01-27:
Crash in Audio output is due to m_worker being a nullpointer:

     AudioOutput::applySettings

        if (m_running) {
            m_worker->setIQMapping(settings.m_iqMapping);
        }

Looks like m_running isn't initialised.

srcejon, 2023-01-27:
There's also a crash when closing Audio Output, looks like the mutex has been deleted, when AudioFifo::write is called.

[image]

Looks like a deleted FIFO in the audioFifos list:

[image]

srcejon, 2023-01-27:
Reopened, as there's a second bug, when trying to close the Audio Output, as shown in the post above.

minexixi, 2023-01-27:
> What OS are you on?
> 
> There isn't an 802.15.4 demodulator. The modulator opens ok for me.
> 
> ATV mod missing is in #1525

Windows 11

minexixi, 2023-01-27:
> 你是什​​么操作系统？
> 
> 没有 802.15.4 解调器。调制器对我来说打开正常。
> 
> 缺少 ATV 模组在#1525

[image]
IEEE 802.15.4 modulator is available most of the time and occasionally results in no response

PR #1575 (2023-01-27): Audio Output: Initialise m_running
Partially fixes #1574
closed 2023-03-25 by github-actions[bot]