f4exb/sdrangel #1579: sdrangel Android size of letters

cvafiadis, 2023-01-31:
Hi to everybody 
thank you for the excellent app
please if possible a sollution for bigger letters otherwise is not workable..

thanks

bubnikv, 2023-02-04:
I have the same issue on Chromebook running the Android build. It works nicely demodulating my KiwiSDR, however the font is extremely tiny, unreadable.

srcejon, 2023-02-04:
> I have the same issue on Chromebook running the Android build. It works nicely demodulating my KiwiSDR, however the font is extremely tiny, unreadable.

Can you send a screenshot please.

bubnikv, 2023-02-04:
[image: image.png]

srcejon, 2023-02-05:
That didn't quite work unfortunately. Can you try attaching the image via the github site rather than email?

bubnikv, 2023-02-06:
[image]

closed 2023-03-25 by github-actions[bot]