f4exb/sdrangel #1581: Crash loading NFM demod

bluzee, 2023-02-01:
Add RTL-SDR receiver, add channel select NFM demod click apply and it crashes everytime.   Version 7.8.6-1 amd64 deb installed into Mint 21 which is Ubuntu 22.04 essentially. 

[New Thread 0x7fff4e9fd640 (LWP 22160)]
2023-01-31 21:05:49.440 (D) NFMDemod::NFMDemod
2023-01-31 21:05:49.441 (D) NFMDemod::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  12500  m_afBandwidth:  3000  m_fmDeviation:  5000  m_volume:  1  m_squelchGate:  5  m_deltaSquelch:  false  m_squelch:  -30  m_ctcssIndex:  0  m_ctcssOn:  false  m_dcsOn:  false  m_dcsCode:  23  m_dcsPositive:  false  m_highPass:  true  m_audioMute:  false  m_audioDeviceName:  "System default device"  m_streamIndex:  0  m_useReverseAPI:  false  m_reverseAPIAddress:  "127.0.0.1"  m_reverseAPIPort:  8888  m_reverseAPIDeviceIndex:  0  m_reverseAPIChannelIndex:  0  force:  true
2023-01-31 21:05:49.441 (D) DSPDeviceSourceEngine::addSink:  NFMDemod
2023-01-31 21:05:49.441 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAddBasebandSampleSink
2023-01-31 21:05:49.441 (D) NFMDemod::start

Thread 1 "sdrangel" received signal SIGILL, Illegal instruction.
0x00007ffff7511d5e in Golay2312::syn(unsigned int*, unsigned int) () from /usr/lib/sdrangel/libsdrbase.so
(gdb) bt
#0  0x00007ffff7511d5e in Golay2312::syn(unsigned int*, unsigned int) () at /usr/lib/sdrangel/libsdrbase.so
#1  0x00007ffff7511f40 in Golay2312::buildCorrMatrix(unsigned char*, unsigned int*, bool) () at /usr/lib/sdrangel/libsdrbase.so
#2  0x00007ffff7512179 in Golay2312::Golay2312() () at /usr/lib/sdrangel/libsdrbase.so
#3  0x00007fff79bf2da4 in DCSDetector::DCSDetector() () at /usr/lib/sdrangel/plugins/libdemodnfm.so
#4  0x00007fff79bfbd85 in NFMDemodSink::NFMDemodSink() () at /usr/lib/sdrangel/plugins/libdemodnfm.so
#5  0x00007fff79c06c79 in NFMDemodBaseband::NFMDemodBaseband() () at /usr/lib/sdrangel/plugins/libdemodnfm.so
#6  0x00007fff79bfc420 in NFMDemod::start() () at /usr/lib/sdrangel/plugins/libdemodnfm.so
#7  0x00007fff79bfcb3b in NFMDemod::NFMDemod(DeviceAPI*) () at /usr/lib/sdrangel/plugins/libdemodnfm.so
#8  0x00007fff79bfcc9b in non-virtual thunk to NFMPlugin::createRxChannel(DeviceAPI*, BasebandSampleSink**, ChannelAPI**) const ()
    at /usr/lib/sdrangel/plugins/libdemodnfm.so
#9  0x00007ffff7dca9fc in MainWindow::channelAddClicked(Workspace*, int, int) () at /usr/lib/sdrangel/libsdrgui.so
#10 0x00007ffff6350793 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007ffff7dab3c5 in DeviceGUI::addChannelEmitted(int) () at /usr/lib/sdrangel/libsdrgui.so
#12 0x00007ffff6350793 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#13 0x00007ffff7dab542 in ChannelAddDialog::addChannel(int) () at /usr/lib/sdrangel/libsdrgui.so
#14 0x00007ffff63507c8 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff6fabf53 in QDialogButtonBox::clicked(QAbstractButton*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#16 0x00007ffff6fac619 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#17 0x00007ffff63507c8 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#18 0x00007ffff6f00116 in QAbstractButton::clicked(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#19 0x00007ffff6f0039e in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#20 0x00007ffff6f01fc4 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#21 0x00007ffff6f021e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging-- 
#22 0x00007ffff6e4d4ee in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#23 0x00007ffff6e0a713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#24 0x00007ffff6e12364 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#25 0x00007ffff6318e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#26 0x00007ffff6e10e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#27 0x00007ffff6e66d40 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#28 0x00007ffff6e69fd5 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#29 0x00007ffff6e0a713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#30 0x00007ffff6318e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#31 0x00007ffff6701307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#32 0x00007ffff66d6a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#33 0x00007ffff1859d6e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#34 0x00007ffff468cd3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#35 0x00007ffff46e16c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#36 0x00007ffff468a3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#37 0x00007ffff63720b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#38 0x00007ffff631775b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#39 0x00007ffff701b4e2 in QDialog::exec() () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#40 0x00007ffff63507c8 in  () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#41 0x00007ffff6f00116 in QAbstractButton::clicked(bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#42 0x00007ffff6f0039e in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#43 0x00007ffff6f01fc4 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#44 0x00007ffff6f021e7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#45 0x00007ffff6e4d4ee in QWidget::event(QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#46 0x00007ffff6e0a713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#47 0x00007ffff6e12364 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#48 0x00007ffff6318e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#49 0x00007ffff6e10e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#50 0x00007ffff6e66d40 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#51 0x00007ffff6e69fd5 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#52 0x00007ffff6e0a713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
#53 0x00007ffff6318e3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#54 0x00007ffff6701307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) ()
    at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#55 0x00007ffff66d6a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
    at /lib/x86_64-linux-gnu/libQt5Gui.so.5
#56 0x00007ffff1859d6e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#57 0x00007ffff468cd3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#58 0x00007ffff46e16c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#59 0x00007ffff468a3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#60 0x00007ffff63720b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#61 0x00007ffff631775b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
--Type <RET> for more, q to quit, c to continue without paging--
#62 0x00007ffff631fcf4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
#63 0x00005555555571f2 in  ()
#64 0x000055555555676b in main ()
(gdb)

f4exb, 2023-02-01:
> Thread 1 "sdrangel" received signal SIGILL, Illegal instruction

Your system does not support SSE 4.2 instructions. Plese check here https://github.com/f4exb/sdrangel/wiki/Hardware-requirements and comments in every release: https://github.com/f4exb/sdrangel/releases/tag/v7.8.6

bluzee, 2023-02-01:
Ok. Thank you. My bad. Sorry.

closed 2023-02-01 by bluzee