f4exb/sdrangel #1582: Cannot find prebuilt *.dmg for MacOS

wolfch, 2023-02-01:
This page: [https://github.com/f4exb/sdrangel/wiki/Quick-start](https://github.com/f4exb/sdrangel/wiki/Quick-start) says:
```
Mac O/S
For most releases you will find a .dmg file.
```
...without any links to such `*.dmg` file(s).  I've been clicking on different links and seem to be going in circles without finding a binary release for MacOS

srcejon, 2023-02-01:
Releases page is: https://github.com/f4exb/sdrangel/releases

Expand the assets section:

[image]

Latest is: https://github.com/f4exb/sdrangel/releases/download/v7.8.6/sdrangel-7.8.6-macx64.dmg

7.9.0 is still building!

closed 2023-03-25 by github-actions[bot]

wolfch, 2023-03-26:
@srcejon - thanks for the link!  That worked. (I missed your initial response, sorry)