f4exb/sdrangel #1585: SDRAngel stops responding after a few days - Windows

davidbelle, 2023-02-03:
Below is the output from the Security and Maintenance window:

```
Source
sdrangel.exe

Summary
Stopped responding and was closed

Date
3/02/2023 11:04

Status
Report sent

Description
A problem caused this program to stop interacting with Windows.
Faulting Application Path:	C:\Program Files\SDRangel\sdrangel.exe

Problem signature
Problem Event Name:	AppHangB1
Application Name:	sdrangel.exe
Application Version:	0.0.0.0
Application Timestamp:	63c1f7d5
Hang Signature:	841a
Hang Type:	134217728
OS Version:	10.0.19045.2.0.0.256.48
Locale ID:	3081
Additional Hang Signature 1:	841aafba9be40db9a82d8e11686232ca
Additional Hang Signature 2:	69b7
Additional Hang Signature 3:	69b7445402a0dbf739916141c84b452f
Additional Hang Signature 4:	841a
Additional Hang Signature 5:	841aafba9be40db9a82d8e11686232ca
Additional Hang Signature 6:	69b7
Additional Hang Signature 7:	69b7445402a0dbf739916141c84b452f

Extra information about the problem
Bucket ID:	a119b79f02a8b6df4ee5287eb30d8344 (2226230115439575876)
```
I've seen this on a few PC's. Computer specs seem powerful enough ranging from 16-32 GB of ram, Intel i7 or i9 3 GHz. 
Average load on the PC is about 50% (also runs other apps but they don't lock up). 

SDRangel is configured with one RTL-SDR

Let me know how I can get any more info if required.

f4exb, 2023-02-03:
Who wants to run SDRangel for days on Windows? It does run for weeks on a Raspberry-Pi ...

davidbelle, 2023-02-03:
While I love to poke fun at Windows, the decision to run this on Windows and not a Raspberry Pi is out of scope / irrelevant to this conversation. If it's suppose to run on windows I assumed you want to know about this issue? It's your software.

f4exb, 2023-02-03:
I will not spend time on such corner cases moreover there is clearly not enough material to investigate. We need at least a stack trace.

davidbelle, 2023-02-03:
The application locks up and becomes unresponsive - there is no stack trace since it doesn't crash. 

> Let me know how I can get any more info if required.

> I will not spend time on such corner cases moreover there is clearly not enough material to investigate. We need at least a stack trace.

Hence why I wrote my comment. I was more than happy to provide more info - just ask. 
Closing this ticket, I'm not interested in wasting my time either.

closed 2023-02-03 by davidbelle