f4exb/sdrangel #1586: Auto saving APT-Images

Oraaaaange, 2023-02-03:
Hello.

First, thanks for this awesome software. I like it alot.

Now my Problem: the APT-Demodulator does not keep the settings for auto saving images. I would like to let run SDRangel for some hours to obtain multiple pictures. But as soon as the satellite tracker starts a record, the setting for auto saving resets so nothing gets saved. It works when i activate auto saving while it is decoding but only until a new record starts. I tried with both windows and linux versions, both beave alike.

Looking forward to a solution or a fix, thank you in advance.

srcejon, 2023-02-03:
Have you created a Device Set Preset with APT-Demod setup with the settings you want, that is loaded by the Sat Tracker on AOS?

Oraaaaange, 2023-02-03:
I did not, now it works. Thanks!

closed 2023-02-04 by srcejon