f4exb/sdrangel #1588: SDRplay not compile. Something missing?

hiperiondev, 2023-02-08:
When compiling sdrangel can't find SDRPLAY:
```
-- SDRPLAY LIBRARIES SDRPLAY_LIBRARIES-NOTFOUND
-- SDRPLAY INCLUDE DIRS SDRPLAY_INCLUDE_DIR-NOTFOUND
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (SDRPLAY)
  does not match the name of the calling package (SDRplay).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/Modules/FindSDRplay.cmake:24 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:758 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find SDRPLAY (missing: SDRPLAY_LIBRARIES SDRPLAY_INCLUDE_DIR) 
```

I actually don't have sdrplay_api.h, which it looks for in cmake/Modules/FindSDRplay.cmake
I didn't find any reference to this on the Wiki, but on github I find: https://github.com/srcejon/sdrplayapi
I don't know if it's relevant.

srcejon, 2023-02-09:
Yes, you need to install the SDRplay API. You should be able to do that with:

git clone https://github.com/srcejon/sdrplayapi.git
cd sdrplayapi
sudo yes | sh install_lib.sh

(Taken from cmake/ci/build_sdrplay.sh)

Give it a try - if it works, we can added to the wiki.

hiperiondev, 2023-02-09:
Yes! With https://github.com/srcejon/sdrplayapi.git compile ok

closed 2023-02-09 by hiperiondev