f4exb/sdrangel #1589: Sonde M10 , DFM

YT3GTI, 2023-02-08:
Hi Edouard , nice program for SDR , probably the best i ever used on my PC !  
Decoded RS41 probes perfect, is there any chance to add M10 , M20 , DFM09 for decoding in program ? 

Thank you very much for this software !

Vlada YT3GTI

f4exb, 2023-02-09:
This plugin was written by @srcejon  Maybe Jon has an idea on how easy, difficult or impossible it is to implement.

srcejon, 2023-02-09:
Only RS41s are used in the UK, so you'd have to upload some .wav IQ files.

YT3GTI, 2023-02-10:
> Only RS41s are used in the UK, so you'd have to upload some .wav IQ files.

@srcejon  thank you for answer , here you have 8 IQ files  for 4 model probes  :   
M10 
https://www.filemail.com/d/whuvmyirdpynnvp

M20 
https://www.filemail.com/d/dxzarunwrbvshin

PS15 DMF : 
https://www.filemail.com/d/eigazymxiidspig

DFM09 : 
https://www.filemail.com/d/noptmquduvpctlu

BW for M10 and M20 is 24000 , and for DFM 10000 

Thnx
Vlada YT3GTI

srcejon, 2023-02-14:
Thanks. Will see if I can find some specs.

f4tnk, 2023-09-06:
Hello,

Votre logiciel est vraiment excellent, bravo et merci beaucoup.

Il me semble que sdrangel est écrit est cpp... il y a le code source de radiosonde_auto_rx (une référence dans le domaine), qui implémente les demodulateurs adéquat M10, M20 (utiliser en france) en cpp...😉

Sa serait super de voir cela dans sdrangel couplé à une map avec le décodage des données de télémétrie...

73's de F4TNK