f4exb/sdrangel #1590: VOR demodulator with ATIS, identifier not stable

keukenrol, 2023-02-09:
Hi

my local VOR has an ATIS broadcast on top of the VOR signal itself. The VOR demodulator has quite a lot of difficulties to get a sustainable result. Usually the beeps are broadcasted a few times after eachother with then 15 seconds pause. During this time the identifier dissapears (timeout issue)? It would be nice to not only set the threshold for the beeps but also the duration and timeout to further tweak the reception.
In the VOR localiser plugin, I never get any data from the demodulator, despite the demodulator sometimes correctly identifying the station.

srcejon, 2023-02-09:
Can you upload a screenshot? (An IQ .wav file with VOR details might also be helpful).

I think there is a bug with the round-robin logic in the localizer (depending how many demods you have and how many VORs you add), but should work for just a single VOR / demod.

keukenrol, 2023-02-09:
I have just recorded an sdriq file. If you can use this, send me a pm and I'll get it to you.

keukenrol, 2023-02-09:
Due to the file size I had to upload it to google drive. Find it here:
https://drive.google.com/file/d/1IrrBnq41CHHf5-dfckTnrTR1BS7OUrbc/view?usp=share_link

srcejon, 2023-02-09:
With regards to ATIS, I don't think it's a timeout issue, just that ATIS is confusing the morse decoder. It might be possible to filter the ATIS audio though,

> In the VOR localiser plugin, I never get any data from the demodulator, despite the demodulator sometimes correctly identifying the station.

Here's a screenshot from the file you send

[image]

The radial is displayed in the VOR Localizer table and on the map, so looks OK. It looks like it's not displaying some of the incorrect morse idents though. (You need to press the play button in the top left of the VOR Localizer window.)

keukenrol, 2023-02-09:
Well, it is quite strange as doing exactly the same does not give me any radial.
[image]

srcejon, 2023-02-09:
You need to double click on the VOR you want the localizer to try to use - and then it will turn green.

keukenrol, 2023-02-09:
Ah well, I might have misinterpreted this. It does work now indeed. I expected the VOR's to be added to the list automatically. Thanks for your help!

f4exb, 2023-02-10:
> I expected the VOR's to be added to the list automatically.

This is something I can look into. Actually I thought it was done... I tried with Jon's recording on sdrangel.org (I do not have good VOR coverage here) and thought it was a bug. Indeed some other features scan available devices at start and react upon adding or removing channels. I think the channel add/delete event was added after the VOR Feature was created.

f4exb, 2023-02-11:
I should read my own documentation sometimes... In fact you click on the VOR (on the map) to add or remove VORs that are watched (thus in the table). Then available VOR channels are used to get and consolidate VOR data. However I noticed a few things funny more noticeably when only one channel is available and you are watching two VORs.

srcejon, 2023-03-06:
The above patch should improve detection of the Morse ident, but unfortunately the voice signal is spread over the Morse signal, so it's not possible to completely remove it with a bandpass filter (You will still be able to hear the odd click if you press the new audio bandpass filter button - Doubling the filter order doesn't really help). If you set MT to about 5, it should mostly decode it correctly (for the IQ file you sent).

srcejon, 2023-03-06:
> (It seems there's a bug if you click it twice, it will be added to the table twice, rather than be removed)

This will only happen if My Position is changed with the Localizer open.

closed 2023-04-21 by github-actions[bot]