f4exb/sdrangel #1592: DSD Demodulator Not Working in SDR Angel

gitnaub, 2023-02-10:
the DSD Demolutaor is not working anymore and I am unable to receive any signals. The blue graphic no longer moves and it seems like the plug is not reading anything. 
I have tried uninstalling and reinstalling the application, but the issue still persists. I believe this is a serious issue that needs to be resolved as soon as possible. The DSD Demolutor is an important feature of the SDR Angel and it not working is causing a lot of inconvenience.

I would appreciate it if the SDR Angel team could investigate this issue and provide a solution. I am confident that this can be fixed quickly and I would be grateful for any assistance that you can provide.

Thank you for your time and attention.

srcejon, 2023-02-10:
Please attached a screenshot showing the problem, your settings and the signal being received.

f4exb, 2023-02-11:
> I would appreciate it if the SDR Angel team could investigate this issue and provide a solution. I am confident that this can be fixed quickly and I would be grateful for any assistance that you can provide. 

The "team" is mostly Jon and myself. Please realize that this is not a commercial product but an Open Source project maintained by enthusiasts. This sort of phrasing does not pertain to this realm.

gitnaub, 2023-02-12:
I wanted to extend my sincerest apologies for the phrasing used in my previous message. I understand that the project you are working on is not a commercial product, but rather an Open Source initiative maintained by enthusiasts. I should have chosen my words more carefully to reflect the collaborative and community-driven nature of the project.
Please accept my apologies for any confusion or misinterpretation , Please accept my apologies for any confusion or misinterpretation that may have resulted from my use of the word "team". It was not my intention to detract from the hard work and dedication that you and Jon have put into this project

gitnaub, 2023-02-12:
[image]
I wanted to bring to your attention an issue I've encountered with the DSD demodulator in the attached file. Unfortunately, it does not seem to be working properly. On the other hand, I'm happy to report that the NFM demodulator is functioning exceptionally well and all the other plugins are working as expected. Thank you for your attention to this matter.

srcejon, 2023-02-12:
Can you please include the device and spectrum GUIs in the screenshot? The more info you give, the more likely someone can help

f4exb, 2023-02-13:
May I draw your attention on this part: 
[image]

f4exb, 2023-02-13:
May I draw your attention on the "Sq" section dials. This is indeed a squelch and it looks like it does not open. First dial is for the squelch level (db)  and second is the attack in ms. I recommend 10ms or 0 to be able to catch the sync sequence as soon as possible. These digital modes are essentially FM and therefore a squelch mechanism can be applied which avoids erratic decodes and useless processing.

closed 2023-03-31 by github-actions[bot]

cborss, 2023-07-20:
Dear Jon, dear Edouard,

I'm having the same issue. Using the IQ demo file from SDRplay (https://sdrplay.com/resources/IQ/dmr.zip) and tuning the DSD Demodulator to 145,190MHz I no longer get properly decoded audio output. I got excellent DSD decoding with SDRangel version 6.20.2. After a longer session of testing this with various SDRangel versions, I can say that this is the latest version where DSD decoding was still working. From version 7.0.0 on, it no longer worked. Do you have an idea what might go wrong?

Best regards,
Christian

[image]
[presets.zip](https://github.com/f4exb/sdrangel/files/12113660/presets.zip)

DO5CA, 2024-01-05:
Hello, the DSD modulator is not displayed on my SDRangel and I cannot select it. Is there a setting or is the program not installed correctly? I've already tried everything I could think of. I just can't get a DSD mod. Can anyone give me help here?

srcejon, 2024-01-05:
> Hello, the DSD modulator is not displayed on my SDRangel and I cannot select it. Is there a setting or is the program not installed correctly? I've already tried everything I could think of. I just can't get a DSD mod. Can anyone give me help here?

Which version of SDRangel are you using? What OS?

Usually this is a problem with one of the shared libraries not loading (Menu Preferences > Log then restart). Have a look in the debug log file to see if they are any relevant messages.

DO5CA, 2024-01-05:
I use version 7.17.3 in several OS Windows 10, Linux mint 20.3 and it is not displayed in DragonOS Focalx either. Am I making a mistake during the installation? or maybe it's my OS?
the error from the LOG "Cannot load library /usr/lib/sdrangel/plugins/libdemoddsd.so: (libmbe.so.1: Kann die Shared-Object-Datei nicht öffnen:"

srcejon, 2024-01-05:
> I use version 7.17.3 in several OS Windows 10, Linux mint 20.3 and it is not displayed in DragonOS Focalx either. Am I making a mistake during the installation? or maybe it's my OS? the error from [...]

You need to separate the problems for each OS. 

What error message do you see on Windows?

On Linux, your problem is libmbe.so.1 can't be found. Did you compile SDRangel yourself or are you using the Ubuntu binary release from github?

What does ls -l /usr/lib/sdrangel/libmbe* show?

DO5CA, 2024-01-05:
Unfortunately, I am not an expert when it comes to correcting operating systems and/or errors. I would need your kind help with what I should do and when.

carsten@carsten-ThinkPad-T510:~$ ls -l /usr/lib/sdrangel/libmbe*
-rw-r--r-- 1 root root   4216 Jan  1 08:37 /usr/lib/sdrangel/libmbe.a
lrwxrwxrwx 1 root root     11 Jan  1 08:37 /usr/lib/sdrangel/libmbe.so -> libmbe.so.1
lrwxrwxrwx 1 root root     13 Jan  1 08:37 /usr/lib/sdrangel/libmbe.so.1 -> libmbe.so.1.3
-rw-r--r-- 1 root root 116944 Jan  1 08:37 /usr/lib/sdrangel/libmbe.so.1.3

DO5CA, 2024-01-05:
I use what Github offers

srcejon, 2024-01-06:
Thanks. 

In a terminal, type:

````
    export LD_LIBRARY_PATH=/usr/lib/sdrangel:$LD_LIBRARY_PATH   
    /usr/bin/sdrangel
````

And then post the log to see if that changes the error message.

DO5CA, 2024-01-09:
Hello,Is there a was to correctly integrate Module or install New ones even after the current Installation?I have the Problem with all OS because, as already written, the DSD module is not integrated, See the log file. And the ADS-B also doesn't work properly for me. Can't get a map.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

srcejon, 2024-01-09:
Did you try the above command?

DO5CA, 2024-01-09:
Ja das habe ich, der Befehl in der Shell hat funktioniert.Aber wie bekomme ich das dauerhaft ohne jeweils über das Terminal zu gehen hin.Es wäre schön, wenn man das ganze irgendwie korrigiert bekäme. Yes, I did, the command in the shell worked. But how do I do this permanently without going through the terminal? It would be nice if the whole thing could be corrected somehow.--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

srcejon, 2024-01-09:
An alternative, is to create a file ```/etc/ld.so.conf.d/sdrangel.conf``` that contains:

```
/usr/lib/sdrangel
```

Then run: ```ldconfig```