f4exb/sdrangel #1593: Nooelec RTL-SDR v5 SDR not appearing in MACOS

simesa, 2023-02-10:
I've got the Nooelec RTL-SDR v5 SDR on my M2 MacBook Air running MAcOS 13.2 and the DMG installer (installed today).

When I add a sampling device it doesn't appear in the list (which has fileimput, KiwiSDR, LocalInput etc.).

If I press the refresh button on that menu the app crashes. 

Cubic, Welle.io and other native Mac apps see it okay. 

Should the Nooelec be supported?

srcejon, 2023-02-10:
It should be.

simesa, 2023-02-10:
Not entirely sure what I'd be looking for but searching the crash report for sdrangel:

Crashed Thread:        39  QThread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 39 Crashed:: QThread
0   ???                           	    0x7ff8902e29a8 ???
1   libsystem_kernel.dylib        	    0x7ff800f4522a __pthread_kill + 10
2   libsystem_pthread.dylib       	    0x7ff800f7cf7b pthread_kill + 263
3   libsystem_c.dylib             	    0x7ff800ec6ca5 abort + 123
4   libc++abi.dylib               	    0x7ff800f37082 abort_message + 241
5   libc++abi.dylib               	    0x7ff800f2823d demangling_terminate_handler() + 266
6   libobjc.A.dylib               	    0x7ff800c25250 _objc_terminate() + 96
7   libc++abi.dylib               	    0x7ff800f364a5 std::__terminate(void (*)()) + 8
8   libc++abi.dylib               	    0x7ff800f36456 std::terminate() + 54
9   QtCore                        	       0x10e99a77c 0x10e978000 + 141180
10  libsystem_pthread.dylib       	    0x7ff800f7d259 _pthread_start + 125
11  libsystem_pthread.dylib       	    0x7ff800f78c7b thread_start + 15

Thread 39 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x0000000000000000  rcx: 0x0000000000000000  rdx: 0x0000000000000000
  rdi: 0x00007ff842a579c0  rsi: 0x0000000000000003  rbp: 0x00007ff800f39f9b  rsp: 0x00000003063e9800
   r8: 0x00000003063ea000   r9: 0x00007ff800f39f9b  r10: 0x0000000000000000  r11: 0x0000000000000000
  r12: 0x000000000001d117  r13: 0x0000003000000008  r14: 0x0000000000000006  r15: 0x0000000000000016
  rip: <unavailable>       rfl: 0x0000000000000283
 tmp0: 0xffffffffffffffff tmp1: 0x00007ff8902e297c tmp2: 0x00007ff800f45220

srcejon, 2023-02-12:
Post the complete log file. You can set the filename under Preferences > Logging menu

closed 2023-03-30 by github-actions[bot]

ntflix, 2023-06-06:
@simesa  did you get anywhere with this? any solution please?