f4exb/sdrangel #1594: RollOff Value for DVB-S2 Decoder

DD0CW, 2023-02-10:
I have a "serious" wish again:

It regards the DVBS2 demodulator.
There is a field called RollOff. 
And it seems as if this is a purely manual input field with default value 0.35.

But it would be better if the RollOff value would be taken from the baseband header
of the incoming DVBS2 Signal.

For example, changing the value from 0.35 to 0.2 increasis the MER by abt. 1 db
when you receive the QO100 beacon, which sends with RollOff 0.2.

Vy 73 de Frank, DDØCW.

f4exb, 2023-02-11:
That's an interesting suggestion I still have to find where it is in the meta data. However I think the "manual" option should be maintained since some payloads other than the QO-100 beacon may not have it. So in fact "auto" should be a selectable option falling back to the "manual" value anyway if this information cannot be retrieved.

DD0CW, 2023-02-11:
https://www.etsi.org/deliver/etsi_en/302300_302399/30230701/01.04.01_20/en_30230701v010401a.pdf
in clause 5.1.6 you will find all information ( 10 bytes header, field MATYPE-2, last 2 bits = Roll-Off factor ).
And it would be great if the read value could also be displayed in binary.