f4exb/sdrangel #1595: SoapySDR is not shown in the Loaded Plugins

jianjianliu0918, 2023-02-11:
I compiled and installed the software according to the guidance document.
The plug-in selected USRP and SOAPYSDR, but in the list of devices, only detected USRP.
The system is Ubuntu 22.04.1lts and the hardware is USRP B210.

[image]
cmake info :
ubuntu:/opt/build/sdrangel/build$ sudo cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo \-DSOAPYSDR_DIR=/opt/install/SoapySDR -DUHD_DIR=/opt/install/uhd -DAPT_DIR=/opt/install/aptdec -DCM256CC_DIR=/opt/install/cm256cc -DDSDCC_DIR=/opt/install/dsdcc -DSERIALDV_DIR=/opt/install/serialdv -DMBE_DIR=/opt/install/mbelib -DCODEC2_DIR=/opt/install/codec2 -DSGP4_DIR=/opt/install/sgp4 -DLIBSIGMF_DIR=/opt/install/libsigmf -DDAB_DIR=/opt/install/libdab -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
-- SDRangel Version: 7.9.0-9-g74dcf31bb
-- Build type set to: RelWithDebInfo
-- Compiling for 24 bit Rx DSP chain
-- Target architecture: x86_64-native
-- Looking for __AVX2__ - forced found
-- Looking for __AVX__ - forced found
-- Looking for __SSE4_2__ - forced found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
-- Interprocedural optimization enabled
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- FAAD LIBRARY /usr/lib/x86_64-linux-gnu/libfaad.so
-- FAAD INCLUDE DIRS /usr/include
-- Checking for module 'sgp4'
--   No package 'sgp4' found
-- SGP4 LIBRARIES /opt/install/sgp4/lib/libsgp4s.so
-- SGP4 INCLUDE DIRS /opt/install/sgp4/include/libsgp4
-- Checking for module 'apt'
--   No package 'apt' found
-- APT LIBRARIES /opt/install/aptdec/lib/libapt.so
-- APT INCLUDE DIRS /opt/install/aptdec/include/apt
-- Checking for module 'libairspy'
--   No package 'libairspy' found
-- libairspy not found.
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
-- libairspyhf not found.
-- Checking for module 'libbladeRF>=2.0'
--   No package 'libbladeRF' found
-- Checking for module 'libhackrf'
--   No package 'libhackrf' found
-- libhackrf not found.
-- Could not find Lime Suite
-- Checking for module 'libiio>=0.7'
--   No package 'libiio' found
-- Checking for module 'libmirisdr'
--   No package 'libmirisdr' found
-- libmirisdr not found.
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- libperseus not found.
-- Checking for module 'librtlsdr'
--   No package 'librtlsdr' found
-- librtlsdr not found.
-- SDRPLAY LIBRARIES SDRPLAY_LIBRARIES-NOTFOUND
-- SDRPLAY INCLUDE DIRS SDRPLAY_INCLUDE_DIR-NOTFOUND
-- Could NOT find SDRPLAY (missing: SDRPLAY_LIBRARIES SDRPLAY_INCLUDE_DIR) 
-- Checking for one of the modules 'soapysdr>=0.4.0;SoapySDR>=0.4.0'
-- Found SoapySDR: /opt/install/SoapySDR/include, /opt/install/SoapySDR/lib/libSoapySDR.so
-- Checking for module 'libxtrx'
--   No package 'libxtrx' found
-- libxtrx not found.
-- Checking for module 'uhd'
--   No package 'uhd' found
-- UHD LIBRARIES /opt/install/uhd/lib/libuhd.so
-- UHD INCLUDE DIRS /opt/install/uhd/include
-- Include demoddatv
-- added optional 'make dev_docs' for developer documentation generation
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- Asking qmake for QT_PLUGINS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/plugins
-- Asking qmake for QT_IMPORTS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/imports
-- Asking qmake for QT_QML_DIR and got /usr/lib/x86_64-linux-gnu/qt5/qml
-- Asking qmake for QT_LIBS_DIR and got /usr/lib/x86_64-linux-gnu
-- CPack generators: 
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/build/sdrangel/build

Can someone give me some advice?

jianjianliu0918, 2023-02-11:
Use the SoapySDRUtil application to detect the device:

ubuntu:/opt/install/SoapySDR/bin$ ./SoapySDRUtil --probe="driver=uhd,type=b200"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=uhd,type=b200
[INFO] [UHD] linux; GNU C++ version 11.3.0; Boost_107400; UHD_4.3.0.0-0-g1f8fd345
[INFO] [b200_impl.cpp:421] [B200] Detected Device: B210
[INFO] [B200] Detected Device: B210
[INFO] [b200_iface.cpp:568] [B200] Loading FPGA image: /opt/install/uhd/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [B200] Loading FPGA image: /opt/install/uhd/share/uhd/images/usrp_b210_fpga.bin...
[INFO] [b200_impl.cpp:468] [B200] Operating over USB 3.
[INFO] [B200] Operating over USB 3.
[INFO] [b200_impl.cpp:523] [B200] Detecting internal GPSDO.... 
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [gps_ctrl.cpp:244] [GPS] No GPSDO found
[INFO] [GPS] No GPSDO found
[INFO] [b200_impl.cpp:619] [B200] Initialize CODEC control...
[INFO] [B200] Initialize CODEC control...
[INFO] [b200_impl.cpp:688] [B200] Initialize Radio control...
[INFO] [B200] Initialize Radio control...
[INFO] [b200_impl.cpp:1099] [B200] Performing register loopback test... 
[INFO] [B200] Performing register loopback test... 
[INFO] [b200_impl.cpp:1108] [B200] Register loopback test passed
[INFO] [B200] Register loopback test passed
[INFO] [b200_impl.cpp:1099] [B200] Performing register loopback test... 
[INFO] [B200] Performing register loopback test... 
[INFO] [b200_impl.cpp:1108] [B200] Register loopback test passed
[INFO] [B200] Register loopback test passed
[INFO] [b200_impl.cpp:815] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [b200_impl.cpp:1149] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [b200_impl.cpp:1162] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [B200] Actually got clock rate 16.000000 MHz.

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=b200
  hardware=B210
  fpga_version=16.0
  fw_version=8.0
  mboard_id=B210
  mboard_name=3000501960665
  mboard_serial=310BR0641
  rx0_antenna=RX2
  rx0_id=Unknown (0xffff)
  rx0_ref_power_key=b2xx_pwr_rx_rx2
  rx0_ref_power_serial=310BR0641#A
  rx0_serial=
  rx0_subdev_name=FE-RX2
  rx0_subdev_spec=A:A A:B
  rx1_antenna=RX2
  rx1_id=Unknown (0xffff)
  rx1_ref_power_key=b2xx_pwr_rx_rx2
  rx1_ref_power_serial=310BR0641#B
  rx1_serial=
  rx1_subdev_name=FE-RX1
  rx1_subdev_spec=A:A A:B
  tx0_antenna=TX/RX
  tx0_id=Unknown (0xffff)
  tx0_ref_power_key=b2xx_pwr_tx_tx+rx
  tx0_ref_power_serial=310BR0641#A
  tx0_serial=
  tx0_subdev_name=FE-TX2
  tx0_subdev_spec=A:A A:B
  tx1_antenna=TX/RX
  tx1_id=Unknown (0xffff)
  tx1_ref_power_key=b2xx_pwr_tx_tx+rx
  tx1_ref_power_serial=310BR0641#B
  tx1_serial=
  tx1_subdev_name=FE-TX1
  tx1_subdev_spec=A:A A:B

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 2 Rx, 2 Tx

srcejon, 2023-02-11:
Did you run sdrangel with:

sdrangel --soapy

jianjianliu0918, 2023-02-13:
> Did you run sdrangel with:
> 
> sdrangel --soapy

Thank you for your reply, your method is useful, it works！

closed 2023-02-13 by jianjianliu0918