f4exb/sdrangel #1596: [Feature idea] Other time signals to radio clock

kord1e, 2023-02-12:
There are other time signals that are not supported by SDRAngel:

RBU (Russia), 66.67 kHz, strong in some parts of Europe

RWM (Russia), several shortwave frequencies (not sure if this provides the actual clock time or just something less)

Chinese and Japanese signals, may be weak in Europe/USA

srcejon, 2023-02-13:
RWM doesn't transmit the time of day.

For support of these (other than BBC), we'd need some IQ .wav files uploaded with several minutes worth of the signals.

kord1e, 2023-02-14:
I can record RBU and also RWM if needed but that's less exciting.

kord1e, 2023-02-14:
Here's a recording of RBU, 12 minutes, started around 20:29 UTC. not exactly centered but close.

https://mega.nz/file/w0BXRZCL#cuCQ9nhoWqEt_ytu23zyBUYVCa2kC7p0lX4ibEItU6I

srcejon, 2023-02-14:
Thanks. Certainly cleaner than I can receive. Looks a bit different to the other signals, so may require some thought :)

srcejon, 2024-04-07:
Have just added JJY. KiwiSDR's are quite useful!

srcejon, 2024-04-07:
Polish signal discussed on the forum: https://sdrangel.org/forum/viewtopic.php?t=142

srcejon, 2026-09-29:
Tried a few Kiwis and can't received Chinese BPC.