f4exb/sdrangel #1598: Compilation problem CMake Error

EvgenRo, 2023-02-14:
When compiling on MacOS, Terminal gives an error:  

CMake Error at /Users/evgen/Qt/Tools/CMake/CMake.app/Contents/share/cmake-3.24/Modules/ExternalProject.cmake:2741 (message):
  error: could not find svn for checkout of lame
Call Stack (most recent call first):
  /Users/evgen/Qt/Tools/CMake/CMake.app/Contents/share/cmake-3.24/Modules/ExternalProject.cmake:4170 (_ep_add_download_command)
  external/CMakeLists.txt:487 (ExternalProject_Add)
  
  How to fix it?
  Thanx.

[image]

srcejon, 2023-02-14:
Looks like you need to install subversion (svn)

srcejon, 2023-02-14:
I don't think svn is part of Qt. It was included with XCode at some point, but I think they removed it.

https://subversion.apache.org/packages.html

EvgenRo, 2023-02-14:
> I don't think svn is part of Qt. It was included with XCode at some point, but I think they removed it.
> 
> https://subversion.apache.org/packages.html

Install SVN from MacPorts - fixed!
Thanx!

closed 2023-02-14 by EvgenRo