f4exb/sdrangel #1599: Compilation problem CMake Error

EvgenRo, 2023-02-15:
After more than an hour of compiling, the Terminal gave such an error:

CMake Error at cmake/Modules/UHDComponent.cmake:59 (message):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:432 (LIBUHD_REGISTER_COMPONENT)
  
  How to fix it?
  
  
[image]

EvgenRo, 2023-02-15:
Fix in CMakeLists.txt  string 432

LIBUHD_REGISTER_COMPONENT("LibUHD" ENABLE_LIBUHD ON "Boost_FOUND;HAVE_PYTHON_MODULE_MAKO" OFF ON)

on

LIBUHD_REGISTER_COMPONENT("LibUHD" ENABLE_LIBUHD ON "Boost_FOUND;HAVE_PYTHON_MODULE_MAKO" ON ON)

f4exb, 2023-02-16:
This is in libuhd code not SDRangel.

closed 2023-02-16 by f4exb