f4exb/sdrangel #1605: MacOSX M1/M2 support

oxoocoffee, 2023-02-23:
Hello

I am just a new user of SDRAngel. Just downloaded the latest "heat map" version and realized it is only x64 version. After installing it should work through an emulator. But when I try to start it I get this message "You can’t open the application “SDRangel” because it may be damaged or incomplete."

I just cloned source and will take a look at it. A quick cmake run produces error

```
CMake Error at cmake/Modules/DetectArchitecture.cmake:193 (message):
  Flag '-march=native' rejected by compiler.  Please adjust ARCH_OPT option.
Call Stack (most recent call first):
  cmake/Modules/CompilerOptions.cmake:3 (include)
  CMakeLists.txt:570 (include)
```

I will take a stab at DetectArchitecture.cmake and add a fix. Any other setup/build  pointers welcome.

73,

satdna, 2023-03-14:
I have the same problem with compilation on Apple SoC (M2). I will appreciate any help.

oxoocoffee, 2023-03-14:
I will try this weekend if I have some time. Else weekend after I will have time to try

oxoocoffee, 2023-03-20:
Ok did make a bit of progress. I have Qt6 installed. Also, all additional packages are installed using [homebrew](https://brew.sh).

```
brew install boost         # 1.81.0_1
brew install opencv      # 4.7.0
brew install codec2      # 1.0.5
brew install hidapi         # 0.13.1
```

I am not worrying about SDRplay for now. Will look at this later.
What is the list of any additional packages that need to be installed and supported version?

I can run cmake with the following arguments
```
cmake  -DENABLE_SDRPLAY=OFF -DCMAKE_MACOSX_RPATH=1 -DENABLE_QT6=1 ..
```

It did produce this output. Can any developer help me out if the above arguments are correct? I like to get it close as possible to actual release builds.

```
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.37.1 (Apple Git-137.1)")
-- SDRangel Version: 7.11.0
-- Build type set to: Release
-- Build for macOS target: local version
-- RPATH support: ON
-- Compiling for 24 bit Rx DSP chain
-- Looking for __x86_64__
-- Looking for __x86_64__ - not found
-- Looking for __i386__
-- Looking for __i386__ - not found
-- Looking for __arm__
-- Looking for __arm__ - not found
-- Looking for __aarch64__
-- Looking for __aarch64__ - found
-- Target architecture: ARM64-native
-- ----------------CMAKE_REQUIRED_FLAGS
-- Looking for __ARM_NEON
-- Looking for __ARM_NEON - found
-- ARM Neon extensions enabled
-- Looking for __AVX512F__
-- Looking for __AVX512F__ - not found
-- Looking for __AVX2__
-- Looking for __AVX2__ - not found
-- Looking for __AVX__
-- Looking for __AVX__ - not found
-- Looking for __SSE4_2__
-- Looking for __SSE4_2__ - not found
-- Looking for __SSE4_1__
-- Looking for __SSE4_1__ - not found
-- Looking for __SSSE3__
-- Looking for __SSSE3__ - not found
-- Looking for __SSE3__
-- Looking for __SSE3__ - not found
-- Looking for __SSE2__
-- Looking for __SSE2__ - not found
-- Interprocedural optimization is disabled for Mac OS
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
[... 222 lines cut ...]
--   Found libavutil, version 57.28.100
-- Checking for module 'libavfilter'
--   Found libavfilter, version 8.44.100
-- Checking for module 'libswscale'
--   Found libswscale, version 6.7.100
-- Checking for module 'libpostproc'
--   Found libpostproc, version 56.6.100
-- Checking for module 'libswresample'
--   Found libswresample, version 4.7.100
-- Found FFmpeg: /opt/homebrew/lib/libavcodec.dylib;/opt/homebrew/lib/libavformat.dylib;/opt/homebrew/lib/libavutil.dylib;/opt/homebrew/lib/libswscale.dylib
-- Include demoddatv
-- Found Doxygen: /opt/homebrew/bin/doxygen (found version "1.9.6") found components: doxygen dot
-- added optional 'make dev_docs' for developer documentation generation
-- Found Dot: /opt/homebrew/bin/dot
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- Asking qmake for QT_PLUGINS_DIR and got /opt/Qt/6.4.3/macos/plugins
-- Asking qmake for QT_IMPORTS_DIR and got **Unknown**
-- Asking qmake for QT_QML_DIR and got /opt/Qt/6.4.3/macos/qml
-- Asking qmake for QT_LIBS_DIR and got /opt/Qt/6.4.3/macos/lib
-- CPack generators:
-- Configuring done (5.8s)
-- Generating done (1.2s)
-- Build files have been written to: /Users/rob/projects/github/sdrangel/build
```

srcejon, 2023-03-20:
Build instructions are here:

https://github.com/f4exb/sdrangel/wiki/Compile-in-MacOS

I wouldn't recommend using Qt 6, yet. None of the maps will work, amongst other things.

oxoocoffee, 2023-03-21:
Would it be possible to create a homebrew bottle package? so it can be easily installed with `brew`?

srcejon, 2023-03-28:
cmake -DENABLE_FUNCUBE=OFF maybe

satdna, 2023-03-28:
Unfortunately, it was **OFF**. Full `cmake` below:

`cmake -Wno-dev -DCMAKE_MACOSX_RPATH=1 -DENABLE_RTLSDR=ON -DENABLE_BLADERF=OFF -DENABLE_AIRSPY=OFF -DENABLE_AIRSPYHF=OFF -DENABLE_FUNCUBE=OFF -DENABLE_HACKRF=ON -DENABLE_IIO=OFF -DENABLE_LIMESUITE=OFF -DENABLE_MIRISDR=OFF -DENABLE_PERSEUS=OFF -DENABLE_SDRPLAY=OFF -DENABLE_SOAPYSDR=ON -DENABLE_XTRX=OFF -DENABLE_USRP=OFF -DRX_SAMPLE_24BIT=ON -DCMAKE_INSTALL_PREFIX=/Users/USer/Dev/sdrangel -DPKG_CONFIG_USE_CMAKE_PREFIX_PATH=TRUE -DBUNDLE=ON -DENABLE_EXTERNAL_LIBRARIES=ON ..`

satdna, 2023-04-06:
I'm trying to bypass this issue:
`[ 63%] Linking CXX shared library ../../../lib/plugins/libmodatv.dylib Undefined symbols for architecture arm64: "__ZN2cv11getTextSizeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEidiPi", referenced from: __ZN12ATVModSource15mixImageAndTextERN2cv3MatE in atvmodsource.cpp.o "__ZN2cv12VideoCapture4openERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEi", referenced from: __ZN12ATVModSource9openVideoERK7QString in atvmodsource.cpp.o "__ZN2cv6imreadERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEi", referenced from: __ZN12ATVModSource9openImageERK7QString in atvmodsource.cpp.o "__ZN2cv7putTextERKNS_17_InputOutputArrayERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEENS_6Point_IiEEidNS_7Scalar_IdEEiib", referenced from: __ZN12ATVModSource15mixImageAndTextERN2cv3MatE in atvmodsource.cpp.o ld: symbol(s) not found for architecture arm64 collect2: error: ld returned 1 exit status make[2]: *** [lib/plugins/libmodatv.dylib] Error 1 make[1]: *** [plugins/channeltx/modatv/CMakeFiles/modatv.dir/all] Error 2 make: *** [all] Error 2`

Is there possibility to bypass compilation of the specific plug-in? _modatv_ in this case?

srcejon, 2023-04-06:
cmake -DENABLE_CHANNELTX_MODDATV=OFF

satdna, 2023-04-08:
Many thanks! Finally I was able to compile the app for M2 without any errors. But one more step is needed.
When I'm starting sdrangel, it's crushing and showing the error:

Exception Type:        EXC_BAD_ACCESS (SIGKILL (Code Signature Invalid))
Exception Codes:       UNKNOWN_0x32 at 0x00000001000a0000
Exception Codes:       0x0000000000000032, 0x00000001000a0000

Termination Reason:    Namespace CODESIGNING, Code 2 Invalid Page

tko, 2023-05-10:
FWIW the following steps seemed to work for me (at least sdrangel starts and few things work)

1. `brew create --set-name sdrangel --cmake https://github.com/f4exb/sdrangel/archive/refs/tags/v7.13.0.tar.gz`
2. add `depends_on "qt5"`
3. `brew install sdrangel`

satdna, 2023-05-27:
> FWIW the following steps seemed to work for me (at least sdrangel starts and few things work)
> 
>     1. `brew create --set-name sdrangel --cmake https://github.com/f4exb/sdrangel/archive/refs/tags/v7.13.0.tar.gz`
> 
>     2. add `depends_on "qt5"`
> 
>     3. `brew install sdrangel`

It's working! Perfect. Many thanks.

haplm, 2023-06-16:
@tko how did you work around the problem with private classes in Qt? I'm getting:
`/tmp/sdrangel-20230616-3484-1lmyd8v/sdrangel-7.13.0/sdrbase/util/osndb.cpp:21:10: fatal error: 'QtGui/private/qzipreader_p.h' file not found`

tko, 2023-06-16:
@haplm I didn't do anything but add the dependency in the cask. The header seems to be where one would expect it to be
```
❯ brew list qt5|grep qzipr
/opt/homebrew/Cellar/qt@5/5.15.9/lib/QtGui.framework/Versions/5/Headers/5.15.9/QtGui/private/qzipreader_p.h
```
🤷

haplm, 2023-06-16:
Right, that was a problem with macports also being installed (because I have used that to compile sdrangel). I have happily uninstalled them, which solved this problem.

Unfortunately now it is crashing without any useful hint like this:
```
[ 56%] Built target inputrtlsdrsrv
make: *** [all] Error 2
```

I need to figure out what this is

haplm, 2023-06-16:
When one creates homebrew formula, it invokes the build like this:
```
cmake -S . -B build
cmake --build build
```

I can start adding compiler arguments after `cmake -S . -B build`, but adding `"-Wno-dev", "-DDEBUG_OUTPUT=ON", "-DENABLE_RTLSDR=ON", "-DENABLE_BLADERF=ON", "-DENABLE_AIRSPY=ON", "-DENABLE_AIRSPYHF=ON", "-DENABLE_FUNCUBE=ON", "-DENABLE_HACKRF=ON", "-DENABLE_IIO=OFF", "-DENABLE_LIMESUITE=ON", "-DENABLE_MIRISDR=ON", "-DENABLE_PERSEUS=ON", "-DENABLE_SDRPLAY=ON", "-DENABLE_SOAPYSDR=OFF", "-DENABLE_XTRX=ON", "-DENABLE_USRP=ON", "-DENABLE_EXTERNAL_LIBRARIES=ON", "-DBUNDLE=ON", "-DRX_SAMPLE_24BIT=ON", "-DCMAKE_BUILD_TYPE=Release"` makes it fail even sooner like this:

```
[ 21%] Completed 'fftw3f'
cd /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E make_directory /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external/CMakeFiles
cd /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E touch /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external/CMakeFiles/fftw3f-complete
cd /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E touch /tmp/sdrangel-20230616-88395-u2v0h7/sdrangel-7.13.0/build/external/fftw3f/src/fftw3f-stamp/fftw3f-done
[ 21%] Built target fftw3f
make: *** [all] Error 2
```

tko, 2023-06-16:
The full log for the failed build step should be in the last file in `~/Library/Logs/Homebrew/sdrangel/`
If you can't immediately recognize the underlying cause and start working on it you should post the whole log for others to spot the error.

That said I'm not sure this issue is the best place to continue as the issues you seem to facing are more around homebrew and general compilation problems, and with your earlier problems due to macports a lot of it might be just because your system is very different from most.

haplm, 2023-06-16:
Thank you.

I have solved the cryptic errors by turning off parallelisation of the build by uncommenting `ENV.deparallelize` in the brew Formula.

This uncovered that  the issues is caused by zlib

```
make[5]: *** No rule to make target `zlib', needed by `lib/libopencv_imgcodecs.4.5.5.dylib'.  Stop.
make[4]: *** [modules/imgcodecs/CMakeFiles/opencv_imgcodecs.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [external/opencv/src/opencv-stamp/opencv-build] Error 2
make[1]: *** [external/CMakeFiles/opencv.dir/all] Error 2
make: *** [all] Error 2
```

I have tried to disable building it `"-DBUILD_ZLIB=OFF"`, but that didn't work.

I agree that there might be an issue with my setup because of the brief period of having macports alongside homebrew, although I have removed macports thoroughly.

vielmetti, 2023-06-25:
using the v7.15.0 release and the home-brew instructions above, I get this

```
==> cmake -S . -B build
Last 15 lines from /Users/emv/Library/Logs/Homebrew/sdrangel/01.cmake:
  Could not find a package configuration file provided by "Qt5Gamepad" with
  any of the following names:

    Qt5GamepadConfig.cmake
    qt5gamepad-config.cmake

  Add the installation prefix of "Qt5Gamepad" to CMAKE_PREFIX_PATH or set
  "Qt5Gamepad_DIR" to a directory containing one of the above files.  If
  "Qt5Gamepad" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:665 (find_package)

-- Configuring incomplete, errors occurred!
```

closed 2023-08-10 by github-actions[bot]