f4exb/sdrangel #1608: Bug in the decoding of the DAB TII?

choupette-change, 2023-02-26:
Hello, SDRangel does not give the correct decoding of the TII of my radio. SDRAngel gives Main TII: FF and Sub TII: OA, another application gives Main TII: 20 and sub TII:10. The official documentation gives 20 and 10. Sincerely
[image]
[image]

srcejon, 2023-02-27:
The sub ID is the same, it's just 0a is in hex and 10 is decimal. Not sure why you are getting FF for the main though.

Can you record a few minutes of IQ data as a .wav file (not .sdriq as that will be double the size), so I can give it a try?

srcejon, 2023-02-27:
Actually, I've found a similar problem here with a Main TII of 20, that is being decoded as ff. The problem appears to be in the DAB library, rather than SDRangel though.

srcejon, 2023-02-27:
> Ok, do you still need a sample of this radio ?

No, it's ok for now, as I've found a transmitter with TII of 20 as well.

> Can you remind me how to record in .wav ? 

Use the file sink, but set filename to file.wav rather than file.sdriq.

choupette-change, 2023-02-27:
OK, I found this site to have the informations on the French DAB transmitters, if it can help you..
https://extranet.arcom.fr/radio/

srcejon, 2023-02-27:
Thanks for the link. 

Does this look right?

[image]

choupette-change, 2023-02-28:
hello, it's ok for me for these 2 examples Thank you

commit de7192b6 (2023-03-04): Update DAB Demod to use latest DAB API, that fixes #1608
closed 2023-03-04 by f4exb