f4exb/sdrangel #1614: Crash in "make" of the latest version ?

choupette-change, 2023-03-05:
Crash in "make " of the latest version ?
[ 37%] Building CXX object pluginssrv/channelrx/demoddsd/CMakeFiles/demoddsdsrv.dir/dsddemodwebapiadapter.cpp.o
[ 38%] Building CXX object pluginssrv/channelrx/radioastronomy/CMakeFiles/radioastronomysrv.dir/radioastronomyworker.cpp.o
[ 38%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/stringrangegui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicitemsettinggui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicargsettinggui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/intervalslidergui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/complexfactorgui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/arginfogui.cpp.o
[ 39%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/sdrgui_autogen/3YJK5W5UP7/qrc_res.cpp.o
[ 39%] Building CXX object pluginssrv/channelrx/demoddsd/CMakeFiles/demoddsdsrv.dir/dsddecoder.cpp.o
[ 39%] Linking CXX shared library ../../../lib/pluginssrv/libradioastronomysrv.so
[ 39%] Built target demodadsbsrv
[ 39%] Linking CXX shared library ../../../lib/pluginssrv/libdemoddsdsrv.so
[ 39%] Built target demodradiosondesrv
[ 39%] Linking CXX shared library ../lib/libsdrgui.so
[ 39%] Built target demodaptsrv
[ 39%] Built target radioastronomysrv
[ 39%] Built target demoddsdsrv
[ 39%] Built target sdrgui
make: *** [Makefile:156 : all] Erreur 2
tutu@tutu-Vivobook-ASUSLaptop-xxxxxxxxxx:/opt/build/sdrangel/build$

srcejon, 2023-03-05:
It’s not a crash, but a compilation error that is probably earlier in the make output. (Running make -j1 may make it easier to see).

 

I would guess that it probably is related to the DAB library update. Make sure you git update / rebuild the DAB library first.

 

Cheers,

Jon

choupette-change, 2023-03-05:
hello, it's ok for the compilation..
I am testing the new DAB demodulator, I have a few remarks:
- ALL transmitters are displayed on 2d and 3d maps, from the start, I thought they were only displayed when pressing the button: "find transmitters on map", this button seems inoperative??
- the display of the 3d map is long and creates slowdowns.
Thank you for this work

srcejon, 2023-03-05:
> * ALL transmitters are displayed on 2d and 3d maps, from the start, I thought they were only displayed when pressing the button: "find transmitters on map", this button seems inoperative??

Yes, they are all displayed, so you can browse them. The button should centre the map on the corresponding transmitter.

It might not if the DAB Ensemble name doesn't match. The database you linked to doesn't appear to have exact names that match what is transmitted. Can you send a screenshot so I can check?

> * the display of the 3d map is long and creates slowdowns.
>   Thank you for this work

It might help if you restrict the number of DAB transmitters that are displayed - use the "Filter Distance" column and enter a value like 75.

Alternatively, you can switch the 3D map off.

closed 2023-04-20 by github-actions[bot]