f4exb/sdrangel #1620: Add channel adds multiple channels

srcejon, 2023-03-07:
Add channel can add multiple channels if multiple devices have been opened in the same workspace.

To reproduce:

- Add test source 1
- Add test source 2
- Close test source 2
- Add AM Demodulator - you should see two AM Demods open, instead of 1

If you open 3 test sources at the start (or other devices), then 3 AM Demods will be opened.

(I would guess connect is probably being called multiple times, resulting in the signal handler being called multiple times).

f4exb, 2023-03-11:
Introduced a few debug messages:
```
2023-03-11 12:02:18.509 (D) ChannelAddDialog::apply: emit add channel
2023-03-11 12:02:18.510 (D) MainWindow::channelAddClicked: workspace: 0x55d08ca597a0 deviceSetIndex: 0 channelPluginIndex: 2
...
2023-03-11 12:02:18.536 (D) AMDemodSink::applySettings:  m_inputFrequencyOffset:  0  m_rfBandwidth:  5000  m_volume:  2  m_squelch:  -40  m_audioMute:  false  m_bandpassEnable:  false  m_audioDeviceName:  "System default device"  m_pll:  false  m_syncAMOperation:  0  force:  false
2023-03-11 12:02:18.536 (D) MainWindow::channelAddClicked: workspace: 0x55d08ca597a0 deviceSetIndex: 0 channelPluginIndex: 2
```
So there is only one signal emitted from the channel add dialog but eventually the `channelAddClicked` slot in the main window is called twice. In fact there is an intermediate step in `DeviceGUI`: https://github.com/f4exb/sdrangel/blob/master/sdrgui/device/devicegui.cpp#L202 

In main window connection is made with the `DeviceGUI::addChannelEmitted` signal.

The issue could come at the remove device step in this loop: https://github.com/f4exb/sdrangel/blob/master/sdrgui/mainwindow.cpp#L1077

f4exb, 2023-03-11:
Commenting out the code apparently solves the problem but the definitive solution is more complex.

f4exb, 2023-03-11:
It may look strange but according to this: https://stackoverflow.com/questions/26553029/how-to-disconnect-a-lambda-function-without-storing-connection you can disconnect the signal by just having `nullptr` as the slot argument. Thus disconnecting prior to connecting solves the issue in the case you delete the second devcie (R1) or the first (R0):
```c++
        QObject::disconnect(
            deviceGUI,
            &DeviceGUI::addChannelEmitted,
            this,
            nullptr
        );
        QObject::connect(
            deviceGUI,
            &DeviceGUI::addChannelEmitted,
            this,
            [=](int channelPluginIndex){ this->channelAddClicked(deviceWorkspace, i, channelPluginIndex); }
        );
```

commit 7928d482 (2023-03-11): MainWindow: disconnect prior to connect in device re-numeration. Fixes #1620
closed 2023-03-11 by f4exb