f4exb/sdrangel #1624: Spectrum initialisation

srcejon, 2023-03-08:
It seems the Spectrum GUI does not completely reflect the spectrum settings when a spectrum is first displayed. To see this:

- Open any device (As an example use the Test Source and enable FM Modulation)
- Press start
- The spectrum will have the phosphor effect, but the phosphor effect button (histogram) will be unchecked.
- If you then press the button, the waterfall share will be adjusted.

This is probably because m_displayHistogram is initialsed to true in GlSpectrumView, but false in SpectrumSettings
Similarly, m_waterfallShare is 0.5 and 0.66. Also, m_displayCurrent is different between the two.

Easily fixed, but which setting would you prefer?

Secondly, how should a channel initialise the SpectrumSettings if it has an embedded spectrum? I see several channels use calls such as (From ft8demodgui):

	ui->glSpectrum->setDisplayMaxHold(true);

However, this only effects GLSpectrum, not GLSpectrumGUI, so the the GUI and SpectrumSettings will not be in sync (E.g. open FT8Demod and max hold is enabled in the spectrum, but the button is not checked).

I see some channels send a SpectrumVis::MsgConfigureSpectrumVis message to SpectrumVis, but SpectrumVis doesn't appear to forward this to GLSpectrumGUI, so doing it like that can also result in inconsistencies.

GLSpectrumGUI has the public functions updateFrom() and updateSettings(), that can result in settings being propagated to both GLSpectrum and SpectrumVis - but these are for the WebAPI, so not as convenient as getSettings()/setSettings() call.

f4exb, 2023-04-08:
Re point #1: the constructor of `GLSpectrumGUI` is supposed to apply its settings to the `GLSpectrum` at constructor time however at this time the `GLSpectrum` is not referenced yet (it is nullptr) so this does not work. Probably this has to take place later.

f4exb, 2023-04-09:
>  so perhaps the settings should just be sent to SpectrumVis, which should then propagate them on to any buddies.

Yes it should work like this

Also the direct calls to `ui->glSpectrum` set commands should simply not be there at least in the case of the FT8 and SSB demods. FT8 demod was copied from SSB demod originally and for some reason this was present in SSB demod but should not be there really.

commit ff6fa973 (2023-04-09): use MsgConfigureSpectrumVis in place of corresponding GLSpectrum set methods. Fixes #1624
commit e384c067 (2023-04-09): use MsgConfigureSpectrumVis in place of corresponding GLSpectrum set methods. Fixes #1624
closed 2023-04-09 by f4exb