f4exb/sdrangel #1625: Compilation failure with Qt 6

Mohamedadlyi, 2023-03-08:
These lines of codes failed to build with Qt 6.2.4 with Core5Compat
https://github.com/f4exb/sdrangel/blob/54288cf4c70e33e0588b3e3f8e6bb68b2b41de7b/sdrbase/maincore.cpp#L361-L362
The error I got is:
```
sdrangel/sdrbase/maincore.cpp:361:60: error: ‘updateTimeout’ is not a member of ‘QGeoPositionInfoSource’
  361 |         connect(m_positionSource, &QGeoPositionInfoSource::updateTimeout, this, &MainCore::positionUpdateTimeout);
      |                                                            ^~~~~~~~~~~~~
sdrangel/sdrbase/maincore.cpp:362:106: error: no match for call to ‘(const QOverload<QGeoPositionInfoSource::Error>) (QGeoPositionInfoSource::Error (QGeoPositionInfoSource::*)() const)’
  362 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |           
```

Iwyil, 2023-03-20:
Hi! Is it to be expected that I am getting the exact first error @Mohamedadlyi also got in this issue?
Unfortunately, I am lost on how to debug or fix it.

First of all, as requested in the wiki, due to lack of knowledge I can't show any evidence of something being wrong in the code. I am commenting only because it is the same error. Feel free to disregard this comment if this is not sufficient (really, I don't want to add noise or drain development time).

QT version: 6.4.3 (platform is ArchLinux)
qt6-5compat is also installed (on ArchLinux, there are no 'core' packages regarding qt)
sdrangel commit: d67aac3b277080fa697b8a0eaa47c0de5f5c0597 (current master, as of 2023-03-20T21:20+00:00)
The build procedure has been as instructed from the wiki
Here is the error. Hope this can help:
```
[ 15%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/maincore.cpp.o
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp: In member function ‘void MainCore::initPosition()’:
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:364:75: error: no match for call to ‘(const QOverload<QGeoPositionInfoSource::Error>) (QGeoPositionInfoSource::Error (QGeoPositionInfoSource::*)() const)’
  364 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

<details>
<summary>Full log of the error</summary>

```
[ 15%] Building CXX object sdrbase/CMakeFiles/sdrbase.dir/maincore.cpp.o
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp: In member function ‘void MainCore::initPosition()’:
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:364:75: error: no match for call to ‘(const QOverload<QGeoPositionInfoSource::Error>) (QGeoPositionInfoSource::Error (QGeoPositionInfoSource::*)() const)’
  364 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/qt6/QtCore/qglobalstatic.h:4,
                 from /usr/include/qt6/QtCore/QGlobalStatic:1,
                 from /home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:18:
/usr/include/qt6/QtCore/qglobal.h:1320:20: note: candidate: ‘template<class R, class T> constexpr decltype (ptr) QConstOverload<Args>::operator()(R (T::*)(Args ...) const) const [with T = R; Args = {QGeoPositionInfoSource::Error}]’
 1320 |     constexpr auto operator()(R (T::*ptr)(Args...) const) const noexcept -> decltype(ptr)
      |                    ^~~~~~~~
/usr/include/qt6/QtCore/qglobal.h:1320:20: note:   template argument deduction/substitution failed:
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:364:75: note:   candidate expects 2 arguments, 1 provided
  364 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/qt6/QtCore/qglobal.h:1308:20: note: candidate: ‘template<class R, class T> constexpr decltype (ptr) QNonConstOverload<Args>::operator()(R (T::*)(Args ...)) const [with T = R; Args = {QGeoPositionInfoSource::Error}]’
 1308 |     constexpr auto operator()(R (T::*ptr)(Args...)) const noexcept -> decltype(ptr)
      |                    ^~~~~~~~
/usr/include/qt6/QtCore/qglobal.h:1308:20: note:   template argument deduction/substitution failed:
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:364:75: note:   types ‘R (T::)(QGeoPositionInfoSource::Error)’ and ‘QGeoPositionInfoSource::Error (QGeoPositionInfoSource::)() const’ have incompatible cv-qualifiers
  364 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/qt6/QtCore/qglobal.h:1337:20: note: candidate: ‘template<class R> constexpr decltype (ptr) QOverload<Args>::operator()(R (*)(Args ...)) const [with Args = {QGeoPositionInfoSource::Error}]’
 1337 |     constexpr auto operator()(R (*ptr)(Args...)) const noexcept -> decltype(ptr)
      |                    ^~~~~~~~
/usr/include/qt6/QtCore/qglobal.h:1337:20: note:   template argument deduction/substitution failed:
/home/damiano/compile/sdrangel/build/sdrangel/sdrbase/maincore.cpp:364:75: note:   mismatched types ‘R (*)(QGeoPositionInfoSource::Error)’ and ‘QGeoPositionInfoSource::Error (QGeoPositionInfoSource::*)() const’
  364 |         connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::error), this, &MainCore::positionError);
      |                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
```
</details>

Given this is the same error as above, before opening a new issue I would like to be sure this is an actual problem, and not one just on my part.
I am open to hints on which might the problem be and how could this be solved (though, since I am inexperienced in c++ and I don't know qt, I would be of little help, I am afraid)

srcejon, 2023-03-20:
I would suggest you use Qt5 (I do!). 6 isn't ready yet.

Iwyil, 2023-03-21:
I have no problem with that. I was trying a new build to debug others things, and thought it might be useful to report this issue being still somewhat present. Anyway, thank you.

Mohamedadlyi, 2023-04-02:
Sorry for late reply , regarding the error that i reported before ,i made a PR fixing it  that was successfully merged , so at first , i suggest you update the project from your side.

Iwyil, 2023-04-05:
I tried to build the qt6 configuration two days ago after a pull, and I am still getting the same error that I reported two weeks ago (even the same line numbers; actually the file from which the errors originated [sdrbase/maincore.cpp] apparently didn't change since my last post), which seems the same as the second error @Mohamedadlyi  got in the first post.

The commit I built is e376dd04541b360e161977d98fa3ed051d27b898 (2023-04-03), which seems to still be the last right now

Mohamedadlyi, 2023-04-09:
sorry i might have missed that one line , i can't tell how it didn't raise any errors after the commit that i have made . i will work on it . but or now you can make on tiny change in that line 
"'
https://github.com/f4exb/sdrangel/blob/54288cf4c70e33e0588b3e3f8e6bb68b2b41de7b/sdrbase/maincore.cpp#L362

""
and replacing it by 
` connect(m_positionSource, qOverload<QGeoPositionInfoSource::Error>(&QGeoPositionInfoSource::errorOccurred), this, &MainCore::positionError); `

PR #1654 (2023-04-10): Fix Qt6 compilation
Fix compilation with Qt 6.4 - Fixes #1625

Note using Qt 6.5.0 fails to build on Windows due to this Qt bug: https://bugreports.qt.io/browse/QTBUG-112204?jql=text%20~%20%22lconvert%20access%20is%20denied%22
closed 2023-04-10 by f4exb