f4exb/sdrangel #1627: Crash on Android when audio comes from DSD

fagci, 2023-03-11:
If i switch off output from DSD, app will not crash.
Log created with crash.
[2023-03-11-13-01-06.txt](https://github.com/f4exb/sdrangel/files/10947814/2023-03-11-13-01-06.txt)

[image]

closed 2023-04-27 by github-actions[bot]

Egghead67, 2023-08-20:
Was this issue ever resolved? All other channels work perfectly on my android tablet- looks fantastic too, but the moment I get a signal while using the DSD Demodulator the app will immediately crash. I've tried several configurations but always have the same issue.

Auxilus, 2023-11-04:
I am still facing this issue, is there any work around for this?