f4exb/sdrangel #1628: hackRF unconfigured

EA3IGB, 2023-03-11:
Hello
When installing drivers with Zadig for an RTL having the HackRF connected, it has been deconfigured. Now when it starts up, only the 3V3 and 1V8 leds go, I put the DFU mode but it is not recognized by the PC either. I have tried something with Hampi (linux) and Pentoo but I still can't find the key that solves the problem. Has anyone had this problem or could help me?  Thanks
[image]
[image]
[image]
[image]

f4exb, 2023-03-11:
Anything related to SDRangel? Maybe you posted this issue in the wrong place...

EA3IGB, 2023-03-11:
NO, I'm just explaining how I got to this, the problem is the Hack, not the SDRAngel

f4exb, 2023-03-11:
OK so I close the issue. The best place to post your question would be in a forum about HackRF. You can still possibly ask for help in the SDRangel forum: http://sdrangel.org/forum/index.php

closed 2023-03-11 by f4exb