f4exb/sdrangel #1631: Compilation error

hanslammerts, 2023-03-17:
Trying to compile on Debian 11, I get the following critical error : 

`[ 33%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/wrappingdatetimeedit.cpp.o
/home/hans/Dev/sdrangel-7.10.0/plugins/channelrx/demoddab/dabdemodsink.cpp: In constructor ‘DABDemodSink::DABDemodSink(DABDemod*)’:
/home/hans/Dev/sdrangel-7.10.0/plugins/channelrx/demoddab/dabdemodsink.cpp:524:30: error: invalid conversion from ‘void (*)(int)’ to ‘tii_data_t’ {aka ‘void (*)(int, void*)’} [-fpermissive]
  524 |     m_api.tii_data_Handler = tiiDataHandler;
      |                              ^~~~~~~~~~~~~~
      |                              |
      |                              void (*)(int)
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/build.make:134: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodsink.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:12854: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/all] Error 2`

Apart from disabling DAB support, how can I fix this error ?

srcejon, 2023-03-17:
git update dab library and rebuild it.

loewal, 2023-03-17:
Compile the DAB-library again.
See manual config.

hanslammerts, 2023-03-17:
That is exactly what I just did.
Threw away the libdab directory and the dab-cmdline directory.
Then did a git clone of the dab-cmdline.git again (and all other commands given in the instructions), so I'm sure I have a fresh install.
Did a make clean, cmake with all the options again.
However...I'm still getting the same error, although in another cpp file (was sdrgui.dir/gui/wrappingdatetimeedit.cpp, now sdrgui.dir/gui/wrappingdatetimeedit.cpp):

`[ 37%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/channel/channelgui.cpp.o
/home/hans/Dev/sdrangel-7.10.0/plugins/channelrx/demoddab/dabdemodsink.cpp: In constructor ‘DABDemodSink::DABDemodSink(DABDemod*)’:
/home/hans/Dev/sdrangel-7.10.0/plugins/channelrx/demoddab/dabdemodsink.cpp:524:30: error: invalid conversion from ‘void (*)(int)’ to ‘tii_data_t’ {aka ‘void (*)(int, void*)’} [-fpermissive]
  524 |     m_api.tii_data_Handler = tiiDataHandler;
      |                              ^~~~~~~~~~~~~~
      |                              |
      |                              void (*)(int)
[ 37%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/feature/featuregui.cpp.o
At global scope:
cc1plus: note: unrecognized command-line option ‘-Wno-inconsistent-missing-override’ may have been intended to silence earlier diagnostics
make[2]: *** [pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/build.make:134: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/dabdemodsink.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:12854: pluginssrv/channelrx/demoddab/CMakeFiles/demoddabsrv.dir/all] Error 2`

Is there something else I can look at ?

srcejon, 2023-03-17:
What exact commands did you use for the git clone? Are you on the right branch?

Did you install dab library in to the place cmake is looking for it?

There error is in the same file - just the build log is a bit confusing as you are doing a parallel build (make -j1 will be clearer - but slower)

hanslammerts, 2023-03-17:
Thanks for the support sofar !

Here's what I did:

My main directory is /home/hans/Dev/sdrangel-7.10.0
In this directory I created a build and install directory

In the case of libdab I did the following in the main directory : 

`git clone https://github.com/srcejon/dab-cmdline
cd dab-cmdline/library
git checkout msvc
mkdir build; cd build
cmake -Wno-dev -DCMAKE_INSTALL_PREFIX=/home/hans/Dev/sdrangel-7.10.0/install/libdab ..
make install`

This created the /home/hans/Dev/sdrangel-7.10.0/install/libdab directory (and contents).

When compiling sdrangel I'm using the following commands : 

`cd /home/hans/Dev/sdrangel-7.10.0/build
cmake -Wno-dev -DAPT_DIR=/home/hans/Dev/sdrangel-7.10.0/install/aptdec -DDAB_DIR=/home/hans/Dev/sdrangel-7.10.0/install/libdab -DCM256CC_DIR=/home/hans/Dev/sdrangel-7.10.0/install/cm256cc -DMBE_DIR=/home/hans/Dev/sdrangel-7.10.0/install/mbelib -DSERIALDV_DIR=/home/hans/Dev/sdrangel-7.10.0/install/serialdv -DDSDCC_DIR=/home/hans/Dev/sdrangel-7.10.0/install/dsdcc -DCODEC2_DIR=/home/hans/Dev/sdrangel-7.10.0/install/codec2 -DSGP4_DIR=/home/hans/Dev/sdrangel-7.10.0/install/sgp4 -DLIBSIGMF_DIR=/home/hans/Dev/sdrangel-7.10.0/install/libsigmf -DCMAKE_INSTALL_PREFIX=/home/hans/Dev/sdrangel-7.10.0/install/sdrangel ..
`

I've been staring at this for some time now, but I cannot see an error....

srcejon, 2023-03-17:
What version of the SDRangel source code do you have? Are you building git head? If not, you should.

hanslammerts, 2023-03-17:
Ok.

I must have gotten the wrong version of sdrangel.
I may have got it from a tar.gz file, instead cloning it from git.

Now that I cloned it from git, everything compiles.
So problem solved.

I now have another issue, but I'll create another issue for that.
Thanks anyway !