f4exb/sdrangel #1633: Hidden window can'

chris451gh, 2023-03-18:


f4exb, 2023-03-18:
> Clicking the eye again, no effect

This is not a possible situation once the channel is hidden the (slashed) eye icon disappears with it.

To recover and show all channels attached to a device click on the "show all channels" button in the bottom bar of the device window: https://github.com/f4exb/sdrangel/blob/master/sdrgui/device/readme.md#b2-show-all-channels

closed 2023-05-04 by github-actions[bot]

Abdull, 2025-03-26:
A short user story (also for me, because I keep on forgetting how to unhide channels):

"Im hearing white noise sound, but it is not coming from the Pager Demodulator channel, which **apparently** is the only added channel...":

[image]

Clicking on "Display sub windows in tabs", **then clicking through all tabs** will reveal which channels are currently hidden:

[image]

After clicking on "Show all channels" (in *each* Rx device window?), all previously hidden channels are shown again, including the *SSB Demodulator* channel making noise and the *ADS-B Demodulator* channel I forgot about running in the background:

[image]